arduino/Arduino #3904: Updating a library does not update the version number right away

matthijskooijman, 2015-10-01:
I had Adafruit CC3000 version 1.0.0 installed, and 1.0.3 has just been released. I went into the library manager, navigated to the library, clicked "Update", which did something and then still showed 1.0.0 as being installed. Only after closing and re-opening the library manager did it show 1.0.3 as installed, so I suspect the version number was not properly updated.

On a related note: I had expected there to be a version dropdown, but instead there was just an "Update" button. I can imagine it's useful to have an update button when you have an outdated version, but does that mean that you can only install a specific version when you're currently on the latest version? That might be cumbersome when you need older versions regularly. Note that I didn't test how this works exactly, I was just wondering.

ffissore, 2015-10-01:
I'll look at the version update issue. As for the dropdown, it's shown only when there is more than one newer version available. When there is just one choice, we left only one button and no dropdown

matthijskooijman, 2015-10-01:
> As for the dropdown, it's shown only when there is more than one newer version available. When there is just one choice, we left only one button and no dropdown

Right. And if there is just one newer version but multiple older ones, is the dropdown also shown?

ffissore, 2015-10-01:
yes

ffissore, 2015-11-17:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

a4jp-com, 2015-12-15:
Maybe this hasn't been fixed or my problem loading 1.0.4 is slightly different.

per1234, 2017-07-04:
@a4jp-com's problem ended up being unrelated to this issue and is explained here: https://github.com/arduino/Arduino/issues/4307#issuecomment-165458422

I just checked and the version number display is working correctly in Arduino IDE 1.8.3 Library Manager so I'm closing this as solved.

closed 2017-07-04 by per1234