arduino/Arduino #3909: Houly build 2-10-2015 4h41 read also .ino files into  folder

pierrotm777, 2015-10-02:
With this new version, the ide read old files saved into folder that exist into my sketch project 
[image]
IDE read also .ino or .pde files found into the folder 'INO en cours' or 'InOut & Doc' .

That was not with the older IDE version !
It's a new feature or an issue ?

ffissore, 2015-10-06:
If this folder is a library, this was fixed since arduino-builder beta22. Please download a fresh hourly http://www.arduino.cc/en/Main/Software#hourly

per1234, 2015-10-07:
@ffissore I believe what pierrotm777 is referring to is demonstrated by this example(using Arduino IDE 1.6.6 2015/10/07 06:43):

folder structure:
```
Blink
|-- Blink.ino
|-- backup
|   |-- Blink1.ino
```
If you compile Blink.ino you get:
```
E:\Stuff\misc\electronics\arduino\temp\Blink\backup\Blink1.ino: In function 'void setup()':

E:\Stuff\misc\electronics\arduino\temp\Blink\backup\Blink1.ino:1:6: error: redefinition of 'void setup()'

 void setup() {

      ^

Blink:1: error: 'void setup()' previously defined here

 void setup() {

      ^

E:\Stuff\misc\electronics\arduino\temp\Blink\backup\Blink1.ino: In function 'void loop()':

E:\Stuff\misc\electronics\arduino\temp\Blink\backup\Blink1.ino:6:6: error: redefinition of 'void loop()'

 void loop() {

      ^

Blink:6: error: 'void loop()' previously defined here

 void loop() {

      ^

exit status 1
'void setup()' previously defined here
```
I'm not sure whether this is a bug or a new feature but if it's a feature then Blink1.ino should be shown as a tab in the IDE.

ffissore, 2015-10-08:
Confirm. Thank you!

ffissore, 2015-10-08:
This issue was moved to arduino/arduino-builder#31

closed 2015-10-08 by ffissore