arduino/Arduino #3910: Uploading to a 32u4 board fails

NicoHood, 2015-10-02:
I tried to upload a program to a pololu 32u4 Mini LV board today and I got the following errors.
Used this PR as build, linux 64 bit.
https://github.com/arduino/Arduino/pull/3896#issuecomment-144994262

Uploading to a leonardo board works as expected. (with HoodLoader2.0.5 but with leonardo PID/VID)

I also tried to install the software files:
https://github.com/pololu/a-star/tree/master/add-on/pololu

Currently trying to find the reason for this bug. However the 2nd error should not happen I guess. I will report the progress here, just that you know this bug exists.

First time hitting upload:
```
avrdude: Version 6.0.1, compiled on Apr 14 2015 at 19:04:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/home/linuxuser/Downloads/arduino-PR-3896-BUILD-410/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/linuxuser/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
avrdude: ser_open(): can't open device "/dev/ttyACM0": Device or resource busy
avrdude: ser_send(): write error: Bad file descriptor
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
```

2nd Time hitting upload:
```
processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:130)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1179)
	at processing.app.Sketch.exportApplet(Sketch.java:1153)
	at processing.app.Sketch.exportApplet(Sketch.java:1125)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2443)
	at java.lang.Thread.run(Thread.java:745)
Caused by: processing.app.SerialException: Error touching serial port '/dev/ttyACM0'.
	at processing.app.Serial.touchForCDCReset(Serial.java:87)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:119)
	... 6 more
Caused by: jssc.SerialPortException: Port name - /dev/ttyACM0; Method name - openPort(); Exception type - Port busy.
	at jssc.SerialPort.openPort(SerialPort.java:164)
	at processing.app.Serial.touchForCDCReset(Serial.java:81)
	... 7 more
```

NicoHood, 2015-10-02:
Its an IDE bug. I recompiled my working HoodLoader2 with their VID/PID. changing the PID (+arduino VID) works but changing both gives the error above. This is "poor" 3rd party support and should be fixed.

DavidEGrayson, 2015-10-02:
I don't think this is a bug in the Arduino IDE.  I suspect you just need to follow the instructions from the [a-star README](https://github.com/pololu/a-star), which say:

> The "udev-rules" directory contains a file named "a-star.rules" for Linux users.
> If you copy this file to /etc/udev/rules.d/, it will tell ModemManager to not
> try to access the virtual serial ports of the A-Star.  This is necessary on some
> systems in order to program the A-Star.

This problem could also be solved by updating ModemManager to a newer verison that includes our [patch](http://cgit.freedesktop.org/ModemManager/ModemManager/commit/?id=09d92ef3338effa85eff7f4efa416afe11f780a0).

NicoHood, 2015-10-05:
Cant the arduino program catch this error? However the exceptions should be fixed.

chrispanag, 2015-10-06:
I have the same bug exactly with my official Arduino Leonardo. I haven't been able to upload any sketches on it (I bought it yesterday)

Firstly I get this:
`avrdude: ser_open(): can't open device "/dev/ttyACM2": Device or resource busy
avrdude: ser_send(): write error: Bad file descriptor`

And then I get this:
`processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:132)
	at processing.app.debug.Compiler.upload(Compiler.java:166)
	at processing.app.Sketch.upload(Sketch.java:1167)
	at processing.app.Sketch.exportApplet(Sketch.java:1141)
	at processing.app.Sketch.exportApplet(Sketch.java:1113)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2380)
	at java.lang.Thread.run(Thread.java:745)
Caused by: processing.app.SerialException: Error touching serial port '/dev/ttyACM2'.
	at processing.app.Serial.touchForCDCReset(Serial.java:92)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:121)
	... 6 more
Caused by: jssc.SerialPortException: Port name - /dev/ttyACM2; Method name - openPort(); Exception type - Port busy.
	at jssc.SerialPort.openPort(SerialPort.java:164)
	at processing.app.Serial.touchForCDCReset(Serial.java:86)
	... 7 more`

NicoHood, 2015-10-06:
Is it from arduino.org or from arduino.cc? Maybe their vendor ID also cause problems. This should be fixed somehow if possible. Or at least a proper error message should be printed.

chrispanag, 2015-10-06:
It is from Arduino.org

matthijskooijman, 2015-10-06:
On other thing to try: Try stopping / killing ModemManager if it is running. I've seen it cause problems by probing all serial ports that *might* be a modem. For a normal Uno, this only causes problems when you upload directly after plugging in, but since a Leonardo resets its USB during upload, this might cause this problem, I believe.

chrispanag, 2015-10-06:
See this: It fixed my problem :)

http://labs.arduino.org/Arduino+IDE+on+Linux-based+OS

matthijskooijman, 2015-10-06:
Ok, closing this then.Thanks for reporting back.

closed 2015-10-06 by matthijskooijman

MortimerNeuss, 2018-03-08:
@chrispanag  Link is down/redirected. The solution to this problem is missing. What have you seen @chrispanag? Please share!

k26dr, 2020-02-13:
I managed to get it to work on Linux by running `sudo arduino` and running it as root.