arduino/Arduino #3911: String class has no member named c_str

gmbandeira, 2015-10-02:
On the 1.6.4 and 1.6.5 versions of the Arduino IDE the String class says to have no member named c_str. I expected this function to work as it does on these softwares when I'm compiling the code to an Arduino Lilypad.

This is the part of the code that it happened and the error that occured:

[image]

Chris--A, 2015-10-03:
I couldn't find the repo for the Galileo.

However you should be able to use the subscript operator to achieve what you need.

Replace `receivedString.c_str()` with `&receivedStrong[0]` and hopefully you'll have some luck. This is of course assuming the Galileo has the subscript operator implemented.

facchinm, 2015-10-14:
https://github.com/01org/corelibs-galileo is the Galileo core official repo, however it seems quite abandoned and issues can't be opened...
Closing the issue as wontfix since we are not in control of Intel's releases

closed 2015-10-14 by facchinm