arduino/Arduino #3912: Request to include TaskScheduler library into standard Arduino IDE

arkhipenko, 2015-10-02:
Hi,

Would you please consider including TaskScheduler library into Arduino IDE set of libraries?
The library is located here: https://github.com/arkhipenko/TaskScheduler.git

I have fixed all the issues outlined in issue #3901, however don't see the library in the latest nightly build.

==============================

TaskScheduler is a light-weight cooperative multitasking library for arduino microcontrollers.
It enables developers to achieve pseudo multi-tasking quite easily, and have additional benefits. 
This library has been used by several people with generally very positive feedback, and extensively used by me.
The library is NOT pre-emptive (and as such requires cooperative programming to be used), does NOT use any of the timers (therefore does not affect PWM pins), and is very stable.

Regards,
Anatoli Arkhipenko

cmaglie, 2015-10-02:
The git revision you tagged as "v1.6.0" doesn't contain a `library.properties` file, as you can see here:
https://github.com/arkhipenko/TaskScheduler/tree/v1.6.0

Probably you have tagged the release **before** committing the library.properties.

Please re-tag the current master (and eventually remove the current "v1.6.0" to keep the repository clean). After fixing this the library will appear automatically within an hour.

arkhipenko, 2015-10-03:
Apologies about the tags. Fixed, pushed and verified. Should be ok now.

ffissore, 2015-10-05:
Duplicate of #3901

closed 2015-10-05 by ffissore