arduino/Arduino #3915: Please follow the freedesktop base dir specs

Goodmind, 2015-10-03:
See: http://standards.freedesktop.org/basedir-spec/basedir-spec-latest.html

ffissore, 2015-10-06:
Can you briefly clarify what we are **not** doing and how we should do it?

Goodmind, 2015-10-06:
Hello,

Place user config directory of Arduino IDE (```~/.arduino15```) in folder that defined by $XDG_CONFIG_HOME or ```~/.config``` (```~/.config/arduino15```)

ffissore, 2015-10-06:
Thank for you suggestion. However, I'm closing this as wontfix. Many apps store their conf in folders other from `.config`. We won't change such an old feature as using `~/.arduino15`, even if not compliant with freedesktop standards

closed 2015-10-06 by ffissore

matthijskooijman, 2015-10-06:
I guess migrating to ~/.config might be useful at some point (perhaps also putting toolchains etc. in ~/.local/lib or so), but it's tricky to get the upgrade path right I'm afraid.

Goodmind, 2015-10-06:
But also many apps use ```~/.config```

Abdillah, 2016-03-15:
@matthijskooijman Is it that hard?

A6GibKm, 2019-08-30:
You can always set `~/.arduino15` as a fallback. No need to deprecate it.

lhanson, 2019-10-17:
In `$CURRENT_CENTURY` there is very little justification for "nah, I dunno what this XDG business is but other programs don't follow it so we won't either". Given the number of thumbs-down votes on @ffissore 's closing the issue with no argument as to why users' preferences shouldn't be respected given that there is zero technical downside, I get the impression I'm not the only one who feels this way.

Would the maintainers reconsider this position? Would a patch respecting the XDG spec be accepted?

lhanson, 2019-10-17:
And, look, @Perlovka has written a patch! https://github.com/arduino/Arduino/issues/7651#issuecomment-538007411

facchinm, 2019-11-29:
@mastrolinux @cmaglie  can we discuss a bit @matthijskooijman 's proposal ?

Thovthe, 2020-04-05:
Doing this makes sense.  I would say lets use @Perlovka's patch but I'm not sure about leaving the file as arduino15, though, I must admit not knowing the exact reason it was called that in the first place.

A6GibKm, 2020-06-21:
For flatpak users it is possible to do 
```
$ flatpak override --nofilesystem=home cc.arduino.arduinoide
$ flatpak override --persist=Arduino cc.arduino.arduinoide # Where you store your files
$ flatpak override --persist=.arduino15 cc.arduino.arduinoide
```
To get rid of `HOME` pollution.

Avamander, 2020-08-01:
This decision should be revisited because it was made five years ago and now is even more outdated than it was back then.

lhanson, 2020-08-03:
@per1234 says in #10486:

> If anyone has unique and productive information to add, they should do so in that issue and the developers can make the decision as to whether it is worth reopening. Fragmenting the discussion across multiple duplicate issue reports is unproductive.

I would argue that the discussion here already contains a fairly complete case for fixing this. There's really nothing more to add other than watching more and more people clicking :+1: on the issue because they're frustrated and clicking :-1: on the flippant comment which justified its initial closure in the first place.

There really isn't any argument *against* fixing this, so at this point I'd say many of us feel that the onus is now on the maintainers to explain why this doesn't even merit reopening.

TheEvilSkeleton, 2020-10-14:
> For flatpak users it is possible to do
> 
> ```
> $ flatpak override --nofilesystem=home cc.arduino.arduinoide
> $ flatpak override --persist=Arduino cc.arduino.arduinoide # Where you store your files
> $ flatpak override --persist=.arduino15 cc.arduino.arduinoide
> ```
> 
> To get rid of `HOME` pollution.

@A6GibKm I appreciate and thank you so much for this. I really appreciate both people like you and Flatpak for finding workarounds to lazy developers' work.

Although keep in mind that this will make the IDE completely restricted, so you will have to import and save files inside `$FLATPAK_APP_ID`. I'd advise also running:

```bash
flatpak override --filesystem=xdg-desktop --filesystem=xdg-documents --filesystem=xdg-downloads --filesystem=xdg-pictures cc.arduino.arduinoide
```

to allow ~/Desktop, ~/Documents, ~/Downloads and ~/Pictures.

r4pr0n, 2021-09-02:
I think this should be revisited, as now the XDG Base Directory Spec is widely adopted and Arduino should support it, too.

rhysday, 2021-09-15:
For every program I use that doesn't follow XDG standard I use a wrapper. So I create XDG_FAKEHOME_DIR that directs to $HOME/.local/data. Then I create a folder there specific to the application. So my wrapper would look like this:
HOME=${XDG_FAKEHOME_DIR}/arduino
exec /usr/bin/arduino "$@"

But this is NOT working with Arduino! It's still creating folders in my HOME, I do not want 3 folders (arduio15, .java and Arduino) cluttering up my system. 

Please fix. Or if someone knows what I need to put in my wrapper.

Perlovka, 2021-11-05:
Hi, folks!

Since my issue (https://github.com/arduino/Arduino/issues/7651)  been locked for comments, I would like to share new patch here.

This patch will split default arduino files location to three categories:
* Config dir - ~/.config/arduino
* Cache dir - all temporary junk will go to ~/.cache/arduino
* Data dir - all installed platforms, libraries etc. will be stored in ~/.local/share/arduino

Note: libraries path changed from Sketch directory to Data dir.

Two stub functions been added to _arduino-core/src/processing/app/Platform.java_ (in addition to existing **getSettingsFolder()** ):
```
  public File getDataFolder() throws Exception {
    return getSettingsFolder();
  }

  public File getCacheFolder() throws Exception {
    return getSettingsFolder();
  }
```
And for Linux system they are overrided in _arduino-core/src/processing/app/linux/Platform.java_ to return proper locations.
All related functions have been changed to use these three functions.

Note: If there is already  ~/.arduino15 directory, no changes will be made, it is needed to move this dir elsewhere to use new locations.

Aplying patch should be safe for Win and Mac users, but this is not tested. To achieve same behaviour as for Linux system, three above functions should be overriden in proper Platform.java file.

The patch: https://github.com/Perlovka/portage-overlay/blob/c7bb6dd35db295d9337dad666a6afb21e631aca5/dev-embedded/arduino/files/arduino-1.8.16-xdg-compliance.patch

**arduino-cli** could be patched as well to use same directories, patch could be found here: https://github.com/arduino/arduino-cli/issues/1538 or in my repo on github.

kalj, 2022-01-31:
It is really sad to see this being ignored by the Arduino team like this. Even when you are getting a patch suggestions and over and over again being explained why this is important.

Saying that you don't have the time / have more important things to do is 100% fine.

Saying that this is not valid is just ignorant.

tworthington, 2022-03-19:
Who died and put the jerks at freedesktop in charge anyway? The change is not needed.

Avamander, 2022-03-19:
@tworthington This change is very needed for Linux on the Desktop to be a nice option, rather than just a tolerable one. I'd suggest you'd appeal to something more constructive than jokes about someone dying.

Not following XDG-BD standard breaks declarative distributions, sandboxing and containerization, backups, roaming home folders, intelligent trash/cache cleanups and by contributing to the mess of nondescript dotfolders in $HOME it also worsens system administration in general. 

So please, do elaborate how this is "not needed" without saying something egotistical like "I don't have a need for those use-cases."

Batwam, 2022-04-16:
I would really like to see this initial mistake to save the config in the wrong folder being reconsidered by the developers. It should be easy to implement ( if there is a ~/.arduino15, propose to transfer to . config/.), clearly matters to users and Arduino IDE is one of the few applications left ignoring this standard.

Kreijstal, 2022-07-21:
Waiting for dev team to reconsider.

TediCreations, 2022-10-02:
> Thank for you suggestion. However, I'm closing this as wontfix. Many apps store their conf in folders other from `.config`. We won't change an old feature such as using `~/.arduino15`, even if not compliant with freedesktop standards

You can always leave the old path as backward compatible. It is really not that hard to satisfy XDG standards while keeping backwards compatibility.