arduino/Arduino #3920: use variables in blink demo sketch

macsimski, 2015-10-04:
A lot of us use a simple blink sketch to check some setup on our bench. Using a variable like the universal LedPin and Blinktime would help this a lot. besides that using variables in demo sketches is a nice introduction in proper coding techniques.

sketch:

/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the Uno and
  Leonardo, it is attached to digital pin 13. If you're unsure what
  pin the on-board LED is connected to on your Arduino model, check
  the documentation at http://www.arduino.cc

  This example code is in the public domain.

  modified: 
  20140508 by Scott Fitzgerald
  20151004 by Simon Claessen  
 */
 
int ledPin = 13;
int waitTime = 1000;

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin 13 as an output.
  pinMode(ledPin, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(ledPin, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(waitTime);              // wait for a second
  digitalWrite(ledPin, LOW);    // turn the LED off by making the voltage LOW
  delay(waitTime);              // wait for a second
}

harryboo, 2015-10-04:
Hi,

dont use a hardcoded pin definition.
Use the LED definition from 'pins_arduino.h' instead.

int ledPin = LED_BUILTIN;

So this will run on all machines.

macsimski, 2015-10-04:
That is even better. I did not know of that definition.

JordanDTaylor, 2015-10-04:
I like this update!  
One small thing, there appears to be a typo in the comment "// LED_BUILDIN is defined...".

NicoHood, 2015-10-05:
I totally agree with this idea. This also shows some very basic programming stuff (like using variables).
I use this example a lot just to see if my device still works and this would speed up things for debugging (because not all boards use pin13 and sometimes I need a faster blink)

harryboo, 2015-10-05:
Why we use an 'int' for this?

We can use the definition 'LED_BUILTIN' within the loop and setup...

matthijskooijman, 2015-10-05:
FWIW, IIRC the 1.0.x version of this example used a variable for the led pin, but apparently this was removed for some reason. Perhaps someone could dig through the git history to see if any reason was documented for removing it?

cmaglie, 2015-10-05:
AFAIR avoiding the use of variables is on purpose: this is a basic example and the concept of "variable" is introduced later.

@tigoe may you confirm?

tigoe, 2015-10-05:
Correct. We try to introduce one concept at a time in the examples, and for absolute beginners, variables are a lesson on their own.

PnsUfuk, 2015-10-07:
Canım Sikerim Belanı beni deli etme kapat aç geçer.

PnsUfuk, 2015-10-07:
Try this: shutdown your arduino and your computer and turn on this shits.

2015-10-07 11:24 GMT+03:00 Ufuk Köksal YÜCEDAĞ <pns.ufuk@gmail.com>:

cmaglie, 2015-10-07:
Closing as wontfix.

@PnsUfuk thanks for your valuable contribution!

closed 2015-10-07 by cmaglie

macsimski, 2015-10-07:
cmaglie: please point to the sketch where "later" lies... I really think that the use of variables is very fundamental. Where better than in a very basic sketch. To be consistent, one should not introduce bad coding practises and correct it later.

cmaglie, 2015-10-07:
I've no strong opinion on this particular matter just followed indication from @tigoe and @shfitz, based on their teaching experience. I'm fine to change the sketch if they are fine with that, so I'll leave them a final answer to your question.

/cc @tigoe @shfitz 

This is the previous edit for the file: 9ad8748d40ad0033abd9431335af93bdf9ca5e4f

tigoe, 2015-10-07:
"bad coding practices" is a matter of opinion. In my experience (18 years teaching this way) I've found that functions and commands are more fundamental than variables, and that introducing them separately helps those students who struggle with math. 

If you prefer to introduce variables in your first class, feel free not to use the examples that are included. But since Arduino was designed for people who struggle with more traditional programming approaches, I would recommend that we leave the examples as they are.

macsimski, 2015-10-07:
The primary reason for changing the sketch to use variables( I like the "digital pin 2 has a pushbutton attached to it. Give it a name" in DigitalReadSerial) is the way you can use the sketch more freely: Let someone change the value of var blinkTime to see instant results (never underestimate the boost in self from such a simple thing like that) and how often I switched back to the blink sketch if something complex i was building did not work and had to change the pin number to the pin I wanted to test is uncountable. c changing the 13 to something onter is uncountable times 3...

per1234, 2015-10-07:
>one should not introduce bad coding practises

In that case why didn't you use const? Of course then you are introducing yet another concept. Where does it end? I agree it's not an ideal sketch to use for testing but that's not the purpose. You can use your own blink sketch for testing.

macsimski, 2015-10-07:
const is optimum, but as you say, it is yet another concept. and where does it end? in emacs of course. ;-)
Actually I already do use my own blink sketch by doing a find replace on the sketch as it now is.

as you have could read from my previous post...

macsimski, 2015-10-07:
btw. this discussion is  starting to look like https://xkcd.com/1205/

and a $ cp ~/rep/blink.ino Arduino/examples/01.Basics/Blink/ once in a while will cost less chars than typed here. 

Sorry to be bothering you all.

tigoe, 2015-10-07:
The question comes up about once a year, either as a forum item or a mailing list item. There hasn't been a strong consensus on it either direction, so I've left it as is.

macsimski, 2015-10-08:
see y'all in a year then... :-(

Warning! Inertia at work.

pwillard, 2015-12-10:
Since "blink without delay" example is more advanced.. why not update *that* sketch with the "preferred" hardware independent variable for built in LED?

shfitz, 2015-12-10:
That example already has pin 13 as a constant, doesn't it?

pwillard, 2015-12-10:
What I'm referring to is:

Replace "const int ledPin =  13;" in blink without delay example with "const int ledPin  =  LED_BUILTIN;"

tigoe, 2015-12-11:
I have no objection to that.