arduino/Arduino #3921: new issue about Due Wire1: DUE as i2c slave:Wire1 connx to a master don't work - for the Mega code works fine

VogonJeltz, 2015-10-04:
new issue about the DUE, this time concerning Wire1:

the followinc worked for the Mega using Wire SDA+SCL.

after this code didn't work ever with the Due, I now tried Wire1 instead (SDA1+SCL1).
But it also doesn't work!

There must be severe bugs in the Due Wire Libs, please fix them  !!

~~~
//  Arduino code to interface an Arduino to an EV3
//  Arduino as an I2C slave
//  compiles for MEGA and DUE, IDE 1.6.5
//  ver. 0.002d

// using Wire1 after Wire didn't work on the Due!
// Wire works fine on Mega !

#include  <SPI.h>
#include  <Wire.h>

//#include <ardustdio.h>

//=====================================================================================
// misc.  
//=====================================================================================

#define  clock()      millis()  

#define  SLAVE_ADDRESS 0x04
#define  MSGSIZE  30
byte     recvarray[MSGSIZE]; 
byte     sendarray[MSGSIZE];

int32_t  flag=0;

//=====================================================================================
//=====================================================================================
void setup() {
   char sbuf[128];   
   int32_t  i=0;

   // Serial terminal window
   i=115200;
   Serial.begin(i);  
   Serial.print("Serial started, baud="); Serial.println(i);

   // Wire1 (i2c)
   Wire1.begin(SLAVE_ADDRESS);  //  start Arduino as a I2C slave, addr=0x04 (7-bit coded)
   Wire1.onReceive(receiveData );
   Wire1.onRequest(sendData );
   memset(sendarray, 0, sizeof(sendarray) );
   memset(recvarray, 0, sizeof(recvarray) );   
   Serial.println("I2C init: done.");

   sprintf(sbuf, "setup(): done.");
   Serial.println(); Serial.println(sbuf);   

}

//=====================================================================================

uint8_t  calcchecksum(uint8_t array[]) {
  int32_t  sum=0;
  for(int i=4; i<MSGSIZE; ++i) sum+=(array[i]);
  return (sum & 0x00ff);
}

//=====================================================================================

void loop()
{  
  char sbuf[128];

  if(flag==1)
   {  
     Serial.println(); Serial.println();

     // do something with the received data
     // and then do something to build the sendarray [4]...[31]
     sendarray[5] += 1;

     sendarray[0] = 0xff;
     sendarray[1] = calcchecksum(sendarray) ;
     sendarray[2] = SLAVE_ADDRESS; //ID check to handle several slaves
     sendarray[3] = flag; // 1= new data; 127=send again request
     //... feddisch!
     flag=0;     
   }
   else
   if(flag==127) {  // builds array with error flag 127
      Serial.println("ERROR! flag==127");
      memset(sendarray, 0, MSGSIZE);
      sendarray[3]=flag;
   }
   sprintf(sbuf, "Sendarr[4]=%4d,    [5]=%4d,    Recvarr[4]=%4d,    [5]=%4d", 
                  sendarray[4], sendarray[5], recvarray[4], recvarray[5]) ;
   Serial.println(sbuf);

   delay(2);  // short break for the cpu and the bus
}

//=====================================================================================

void receiveData(int byteCount) {
    int32_t i;
    byte val;

    while(Wire1.available()<MSGSIZE) ; // wait for 30  bytes to complete
    i=0;  // init counter var
    while(Wire1.available()&& (i<MSGSIZE) )         // read all bytes
    {
      val=Wire1.read();
      recvarray[i++]=val;
    }
    // check for first 3 bytes and via calcchecksum() function
    if( (recvarray[0]==0xff) && (recvarray[1]==calcchecksum(recvarray)) && (SLAVE_ADDRESS==recvarray[2]) )    
    // if ok:
       flag=1;        // data ok
    else flag=127; // else handle receive error => flag =127  to send array back, request to send again
}

//=====================================================================================

void sendData(){
  // Wire1.write writes data from a slave device in response to a request from a master
  Wire1.write(sendarray, MSGSIZE); // send own 30 bytes back to master..
}

//=====================================================================================
~~~

VogonJeltz, 2015-10-07:
verified by another Due and another mega:

For the other Mega it works, too,
***for another DUE it also doesn't work, neither by Wire nor by Wire1.***

VogonJeltz, 2015-10-21:
now  how to disable the internal pullups on first i2c port?

sandeepmistry, 2015-11-04:
Duplicate of #4007

closed 2015-11-04 by sandeepmistry