arduino/Arduino #3923: Code uploading on DUE is not working in UBUNTU

Atirma, 2015-10-06:
Hi,
We're facing exactly the problem declared in this link.
http://askubuntu.com/questions/681555/ubuntu-14-04-arduino-due-upload-firmware-issue
This problem is serious as it renders the DUE board useless under UBUNTU.

NicoHood, 2015-10-06:
Why dont you switch to the 16u2 usb port?

ffissore, 2015-10-07:
@Fede85 is having the same issue with his ubuntu 14.04, while I'm not with mine. My only guess so far it's that it's hardware related. Can you spare a 14.04 on a different pc and retest?

Atirma, 2015-10-07:
Hi ffissore,
We have tested this issues on two PCs. Luckily, one exhibits this problem, the other doesn't.
Using command 'lsb_release --all' we obtain the following with PC that is able of uploading to the DUE:
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 14.04.1 LTS
Release:        14.04
Codename:       trusty

Executing the same command with the PC where bossac fails, we get:

No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 14.04.3 LTS
Release:        14.04
Codename:       trusty

Main difference is that the first one is Ubuntu 14.04.1 LTS  while the second one is Ubuntu 14.04.3 LTS

'uname -a' returns:

3.13.0-32-generic #57-Ubuntu SMP Tue Jul 15 03:51:08 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux for Ubuntu 14.04.1 LTS, and

3.13.0-65-generic #105-Ubuntu SMP Mon Sep 21 18:50:58 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux for Ubuntu 14.04.3 LTS

I'm attaching also a couple of log files detailing the modules loaded in both cases.
Regards,

------- For the PC where bossac works ---------------------------------------------------------------------------
Script iniciado (miÃ© 07 oct 2015 12:24:41 WEST
)[38;5;202m[38;5;245mantodom[00m@[38;5;172mdis172ac[00m:[38;5;5m~[00mâŸ« vim EKD_time.tex exit[Kgit push origin mastercommit -a -m "cambios"status[Kcd git-projects/siani/
[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[Cdmesg[Kmake upload[6Pdmesgcd git-projects/siani/
[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[12Pgit statuscommit -a -m "cambios"[4Ppush origin master
[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[C[Cexit[Kvim EKD_time.tex [Klsmod
Module                  Size  Used by
cdc_acm                28803  0 
ipt_MASQUERADE         12880  3 
iptable_nat            13011  1 
nf_nat_ipv4            13263  1 iptable_nat
nf_nat                 21841  3 ipt_MASQUERADE,nf_nat_ipv4,iptable_nat
nf_conntrack_ipv4      15012  2 
nf_defrag_ipv4         12758  1 nf_conntrack_ipv4
xt_conntrack           12760  1 
nf_conntrack           96976  6 ipt_MASQUERADE,nf_nat,nf_nat_ipv4,xt_conntrack,iptable_nat,nf_conntrack_ipv4
ipt_REJECT             12541  2 
xt_CHECKSUM            12549  1 
iptable_mangle         12695  1 
xt_tcpudp              12884  6 
bridge                110793  0 
stp                    12976  1 bridge
llc                    14552  2 stp,bridge
ip6table_filter        12815  0 
ip6_tables             27025  1 ip6table_filter
iptable_filter         12810  1 
ip_tables              27239  3 iptable_filter,iptable_mangle,iptable_nat
ebtable_nat            12807  0 
ebtables               30913  1 ebtable_nat
x_tables               34059  11 ip6table_filter,xt_CHECKSUM,ip_tables,xt_tcpudp,ipt_MASQUERADE,xt_conntrack,iptable_filter,ebtables,ipt_REJECT,iptable_mangle,ip6_tables
bnep                   19624  2 
rfcomm                 69160  0 
bluetooth             391196  10 bnep,rfcomm
binfmt_misc            17468  1 
nls_iso8859_1          12713  1 
snd_hda_codec_hdmi     46254  4 
joydev                 17381  0 
snd_hda_codec_via      27860  1 
intel_rapl             18773  0 
x86_pkg_temp_thermal    14205  0 
intel_powerclamp       14705  0 
coretemp               13435  0 
[... 235 lines cut ...]
antodom@dis172acp:~$ uname -a
Linux dis172acp 3.13.0-65-generic #105-Ubuntu SMP Mon Sep 21 18:50:58 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
antodom@dis172acp:~$ lsb_release --all
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.3 LTS
Release:	14.04
Codename:	trusty
antodom@dis172acp:~$ dpkg -i[Kl | grep usb
ii  cups-filters-ipp[01;31m[Kusb[m[Kxd                                 1.0.52-0ubuntu1.3                                   amd64        ipp[01;31m[Kusb[m[Kxd - Daemon for IPP-over-USB printer support
ii  libg[01;31m[Kusb[m[K2:amd64                                        0.1.6-5                                             amd64        GLib wrapper around lib[01;31m[Kusb[m[K1
ii  lib[01;31m[Kusb[m[K-0.1-4:amd64                                    2:0.1.12-23.3ubuntu1                                amd64        userspace USB programming library
ii  lib[01;31m[Kusb[m[K-1.0-0:amd64                                    2:1.0.17-1ubuntu2                                   amd64        userspace USB programming library
ii  lib[01;31m[Kusb[m[K-1.0-0:i386                                     2:1.0.17-1ubuntu2                                   i386         userspace USB programming library
ii  lib[01;31m[Kusb[m[K-1.0-0-dev:amd64                                2:1.0.17-1ubuntu2                                   amd64        userspace USB programming library development files
ii  lib[01;31m[Kusb[m[K-1.0-doc                                        2:1.0.17-1ubuntu2                                   all          documentation for userspace USB programming
ii  lib[01;31m[Kusb[m[K-dev                                            2:0.1.12-23.3ubuntu1                                amd64        userspace USB programming library development files
ii  lib[01;31m[Kusb[m[Kmuxd2                                           1.0.8-2ubuntu1                                      amd64        USB multiplexor daemon for iPhone and iPod Touch devices - library
[... cut ...]
ii  [01;31m[Kusb[m[Kmuxd                                               1.0.8-2ubuntu1                                      amd64        USB multiplexor daemon for iPhone and iPod Touch devices
ii  [01;31m[Kusb[m[Kutils                                              1:007-2ubuntu1                                      amd64        Linux USB utilities
ii  xserver-xorg-video-sis[01;31m[Kusb[m[K                             1:0.9.6-2build1                                     amd64        X.Org X server -- SiS USB display driver

Atirma, 2015-10-12:
Hi, 
Just a quick update/hint on this issue. We had tested that everything is fine until kernel 3.13.0-64-generic is installed (i.e. 3.13.0-63-generic is the last kernel where this problem does not show up). 

We have tested this booting the system through the "Advanced options for Ubuntu" in the grub booting screen and selecting the adequate kernel image. This is also an EMERGENCY shortcut to deal with this problem.

This issue is clearly related to reading data from a USB device. While dealing with these tests, we found also an unexpected problem  with a USB XBee adapter where outgoing messages were sent without problem but incoming messages were not received (in esence the same problem with bossac). This problem was solved also if a 3.13.0-63-generic (or previous) kernel was used.

Regards

NicoHood, 2015-10-12:
What usb port are you using? The one connected to the DUE or the one connected to the 16u2.
If its the 2nd, then its probably a delay problem. the chip erase and reset timing might be wrong due to a lagg. USB2 or 3 used here? with USB HUB and which version (2.0 HUB or 3.0)?

Atirma, 2015-10-12:
Hi, 
This problem occurs independently of the port used. They were connected to a USB 2.0 port.
Regards

NicoHood, 2015-10-13:
You mean on the DUE USB port and the 16u2 USB Port? (you know your arduino has 2 usb plugs).
Have you tested something like an uno or leonardo too?

Atirma, 2015-10-13:
Yes, we found this problem with the two DUE's port (the programming port and the native port). However, we haven't tested other boards yet.

NicoHood, 2015-10-13:
Other boards would be nice to test. I only have AVR boards, but I programmed/modified a lot of the bootloaders. So I could play with this under my linux (where everything runs fine though).

facchinm, 2015-10-13:
The changelog between 3.13.0-63 and 3.13.0-64 (https://launchpad.net/ubuntu/+source/linux/3.13.0-64.104) shows that a bunch of modifications has been made for both the tty and the usb subsystems.
Particularly, I'm afraid that this two commits can be related with the problem:
```
  * xhci: Calculate old endpoints correctly on device reset
    - LP: #1493305
  * n_tty: Refactor input_available_p() by call site
    - LP: #1397976
```
However we can only report this issue to Ubuntu maintainers as a generic bug until we are able to bisect the offending commit

facchinm, 2015-10-13:
We are not alone :smile: https://bugreports.qt.io/browse/QTBUG-48499

facchinm, 2015-10-13:
Reported [here] (https://bugs.launchpad.net/ubuntu/+source/linux-lts-trusty/+bug/1500860)

matthijskooijman, 2015-10-13:
I'm running a Debian machine with kernel 4.0.2, which can upload to the Due (both ports) just fine (just tested with Arduino IDE 1.6.5). Perhaps this is a kernel bug that has since been fixed and we shouldn't spend to much time on this? Perhaps the original reporter can try a newer kernel version (from a newer Ubuntu version?) and see if that helps?

facchinm, 2015-10-13:
@matthijskooijman , I'm running kernel 4.2 on Arch so I should be even less concerned :smile: 
The problem here is that 3.13 kernel has been chosen by Ubuntu as LTS kernel for 14.04 (so support will end in 2017) and ubuntu.com/download/desktop points to that release...

NicoHood, 2015-10-13:
I am running kernel:
Linux linuxpc 3.16.0-50-generic #67~14.04.1-Ubuntu SMP Fri Oct 2 22:07:51 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

I have no due, but all other avr devices work. Using Elementary OS based on Ubuntu 14 LTS

Atirma, 2015-10-14:
Hi,
Good news. It looks like this problem is solved in  the latest kernel (3.13.0-66-generic) update. 
Regards

facchinm, 2015-10-14:
:+1: :+1: 
https://launchpad.net/ubuntu/+source/linux/3.13.0-66.107 shows it was solved by
* n_tty: Fix poll() when TIME_CHAR and MIN_CHAR == 0
    - LP: #1397976

closed 2015-10-14 by facchinm