arduino/Arduino #3925: another bug about DUE Wire1

VogonJeltz, 2015-10-06:
just encountered:
for DUE-master to DUE-slave  connected by i2c, the following code works, if both are running Wire (SDA+SCL).

But connx fails, if one is using Wire1 (SDA1+SCL1)
WT* is going wrong with  Wire1 ??

Master:
~~~
#include <Wire.h>

void setup()
{
  Wire1.begin();
  Serial.begin(9600);
}
  
void loop()
{
  byte address=0x08;
  byte x=100;

  Wire1.beginTransmission(address);
  Wire1.write(x);
  Wire1.endTransmission();
  delay(100);
}
~~~
Slave:
~~~
#include <Wire.h>

byte x;

void setup() {
  Serial.begin(9600);
  Wire.onReceive(empfangen);
  Wire.begin(8);
}

void loop() {
  Serial.println(x);
  delay(100);
}

void empfangen (int numBytes)
{
  while(Wire.available())
  {
   x = Wire.read();
  }
}
~~~

VogonJeltz, 2015-10-07:
error could not be reproduced reliably  using Wire1, maybe also a Due hardware failure.

VogonJeltz, 2015-10-07:
update: 
this issue for Due to Due was just temporary, now it works.
(Other issue about EV3-master to Due-salve  don't work nevertheless constantly)

closed 2015-10-07 by VogonJeltz

sandeepmistry, 2015-10-07:
@VogonJeltz things look ok to me, I tried out your sketches on two Due boards.

Wiring as follows:
 1. Connected grounds together
 1. Due Master SCL1 <-> Due Slave SCL0
 1. Due Master SDA1 <-> Due Slave SDA0

When I open up the serial monitor on the slave, 100 is continuously printed.

Can you please confirm you hardware wiring?

VogonJeltz, 2015-10-08:
I did, and I'm not sure why it didn't work right from the start. Wiring was always cross-checked. Maybe a intermittent contact failure in the pin header contact though.