arduino/Arduino #3928: Arduino IDE serial ports not working on EL Capitan OSX

HBRancher, 2015-10-07:
Anyone know how to fix Arduino IDE serial port lockup on EL Capitan OSX ??
Thanks

matthijskooijman, 2015-10-07:
You are the first to report this issue, so could you please provide more details? What happens exactly? What did you expect? Any error messages / verbose output you could add?

wizardofwords, 2015-10-13:
Hi Matthijs Kooijman,

I have the very same problem.

Arduino IDE does no longer show "tty" ports in port list, but just "cu", even if device "/dev/tty.usbmodem1A21" looks to be correctly created when USB cable is connected.
Second, when I try to transfer any sketch, after compilation and link, when it tries to transfer datas to my genuine Arduino UNO, process gets stuck completely and no return error is reported.
My bluetooth mouse also stacks, while, fortunately, touch pad still works, so I can close Arduino IDE. My Mac fan, that usually can be barely heard, starts to slowly to increase its speed till maximum.

If I run again Arduino IDE in this situation, it reports no USB port at all, even if I disconnect and reconnect my Arduino UNO.

The only way to stop these effects is to turn off and back on my Mac.

I hope I gave you enough elements to start to catch this issue, that has completely stacked my project.

I look forward you hearing from you.

Best Regards.

Luca Moscardini

facchinm, 2015-10-14:
Note to self:
see https://github.com/voodootikigod/node-serialport/pull/607 for a clever fix

wizardofwords, 2015-10-18:
> Note to self:
> see voodootikigod/node-serialport#607 <https://github.com/voodootikigod/node-serialport/pull/607> for a clever fix

Hi Martino,

Unfortunately I had no success: “nam install tmpvar/node-serialport#el-capitan” returns a long list of errors (see below), so I still can’t fix that issue.

Please also find attached a copy of the file “npm-debug.log”.

I look forward to hearing from you.

Best Regards.

Luca Moscardini

npm ERR! git rev-list -n1 el-capitan: fatal: ambiguous argument 'el-capitan': unknown revision or path not in the working tree.
npm ERR! git rev-list -n1 el-capitan: Use '--' to separate paths from revisions, like this:
npm ERR! git rev-list -n1 el-capitan: 'git <command> [<revision>...] -- [<file>...]'
npm ERR! git rev-list -n1 el-capitan: ls -l
npm ERR! git rev-list -n1 el-capitan: fatal: ambiguous argument 'el-capitan': unknown revision or path not in the working tree.
npm ERR! git rev-list -n1 el-capitan: Use '--' to separate paths from revisions, like this:
npm ERR! git rev-list -n1 el-capitan: 'git <command> [<revision>...] -- [<file>...]'
npm ERR! git rev-list -n1 el-capitan: 
npm ERR! git clone --template=/Users/luca/.npm/_git-remotes/_templates --mirror git@github.com:tmpvar/node-serialport.git /Users/luca/.npm/_git-remotes/git-github-com-tmpvar-node-serialport-git-bde745f9: Cloning into bare repository '/Users/luca/.npm/_git-remotes/git-github-com-tmpvar-node-serialport-git-bde745f9'...
npm ERR! git clone --template=/Users/luca/.npm/_git-remotes/_templates --mirror git@github.com:tmpvar/node-serialport.git /Users/luca/.npm/_git-remotes/git-github-com-tmpvar-node-serialport-git-bde745f9: Permission denied (publickey).
npm ERR! git clone --template=/Users/luca/.npm/_git-remotes/_templates --mirror git@github.com:tmpvar/node-serialport.git /Users/luca/.npm/_git-remotes/git-github-com-tmpvar-node-serialport-git-bde745f9: fatal: Could not read from remote repository.
npm ERR! git clone --template=/Users/luca/.npm/_git-remotes/_templates --mirror git@github.com:tmpvar/node-serialport.git /Users/luca/.npm/_git-remotes/git-github-com-tmpvar-node-serialport-git-bde745f9: 
npm ERR! git clone --template=/Users/luca/.npm/_git-remotes/_templates --mirror git@github.com:tmpvar/node-serialport.git /Users/luca/.npm/_git-remotes/git-github-com-tmpvar-node-serialport-git-bde745f9: Please make sure you have the correct access rights
npm ERR! git clone --template=/Users/luca/.npm/_git-remotes/_templates --mirror git@github.com:tmpvar/node-serialport.git /Users/luca/.npm/_git-remotes/git-github-com-tmpvar-node-serialport-git-bde745f9: and the repository exists.
npm ERR! Darwin 15.0.0
npm ERR! argv "/usr/local/Cellar/node/4.2.1/bin/node" "/usr/local/bin/npm" "install" "tmpvar/node-serialport#el-capitan"
npm ERR! node v4.2.1
npm ERR! npm  v2.14.7
npm ERR! code 128

npm ERR! Command failed: git clone --template=/Users/luca/.npm/_git-remotes/_templates --mirror git@github.com:tmpvar/node-serialport.git /Users/luca/.npm/_git-remotes/git-github-com-tmpvar-node-serialport-git-bde745f9
npm ERR! Cloning into bare repository '/Users/luca/.npm/_git-remotes/git-github-com-tmpvar-node-serialport-git-bde745f9'...
npm ERR! Permission denied (publickey).
npm ERR! fatal: Could not read from remote repository.
npm ERR! 
npm ERR! Please make sure you have the correct access rights
npm ERR! and the repository exists.
npm ERR! 
npm ERR! 
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/luca/particle-cli/npm-debug.log

facchinm, 2015-10-20:
Sorry Luca, my message was intended as a reminder for myself, not a real way to solve the issue.
We'll tackle is very soon I promise :smile:

wizardofwords, 2015-10-20:
Hi Martino,
Thanks so much for your time, I really appreciate that.
I would greatly apreciate also if you could recommend even a temporary solution to that issue.
Thanking you in advance, I look forward to hearing from you.
Kind Regards.

Luca Moscardini

-------- Messaggio originale --------

Data: 20/10/2015  09:57  (GMT+01:00) 
A: arduino/Arduino <Arduino@noreply.github.com> 
Cc: wizardofwords <wizardofwords@gmail.com> 
Oggetto: Re: [Arduino] Arduino IDE serial ports not working on EL Capitan OSX  (#3928)

HBRancher, 2015-10-20:
Unfortunately, he only fix that I have found, so far, is the return to Yosemite....  I've reported the problem to Apple... Under El Capitan, any use of the USB ports would lockup my Bluetooth keyboard and mouse.  So, either the drivers from the Arduino must be re-written.... or... there is actually a bug in El Capitan....
Good Luck to all.... I'm also still waiting for any solutions...

tmpvar, 2015-10-21:
Heads up, the pull request @facchinm mentioned has been landed and is included in serialport@2.0.1

`npm install serialport@latest` should bring in these changes if you (@wizardofwords) wanted to try again!

michaelemery, 2015-10-21:
This issue is not a bug, but rather the result of Apple's new System Integrity Protection (SIP) policy called "Rootless". Rootless was introduced with the launch of El Capitan. The effect of Rootless is that many applications and drivers (such as those required for your USB serial connections) will no longer operate correctly. Third party vendors must modify their drivers and applications in order to comply with the new policy.

There are "fixes" floating around the internet that require you to manually disable certain El Capitan security protection features – it is not recommended that you implement these fixes as they are either dangerous, or opportunistic security scams. The only thing you can really do is wait for third party vendors to rewrite their software.

Unfortunately, this is not the answer anyone wanted to hear.

Dark-Archer, 2015-10-25:
You say that 'Third Party Vendors' need to re-write their software, Does this mean that genuine Arduino boards will still work?

ffissore, 2015-10-26:
Fixed by #4006. Available with latest hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-10-26 by ffissore

cmaglie, 2015-10-26:
For the record: we tried the fix on our macs with various Arduino/Genuino boards and it is working flawlessly. If you are still experiencing problems maybe your boards uses proprietary USB-Serial chips that are not compliant with CDC-ACM?

michaelemery, 2015-10-27:
@cmaglie - are you using El Capitan? I had no problems with Yosemite on either of my macs (including my MacBook 2015 with USB C). However, since installing El Capitan on both my macs, the USB serial ports will only communicate using Windows 10 as a Bootcamp OS. Earlier versions of Windows would be fine too, but they have to be run natively (not as a virtual OS within El Capitan). If you are using El Capitan, then you must be using newer drivers.

cmaglie, 2015-10-27:
> are you using El Capitan?

yes, the issue is all about "El Capitan".

> I had no problems with Yosemite on either of my macs (including my MacBook 2015 with USB C). However, since installing El Capitan on both my macs, the USB serial ports will only communicate using Windows 10

Which boards are you using? In particular, I want to know if the USB-2-Serial chip is an atmega16u2 with Arduino's firmware or is another proprietary chip like FTDI or CH340.

wizardofwords, 2015-10-27:
Hi Federico,

I download latest nightly release but it does not fix my Mac problem  with USB ports.

Regards.

Luca Moscardini

PaulStoffregen, 2015-10-28:
Are you certain you're using a genuine Arduino board?  Not one of those Chinese clones with a CH340 chip?

wizardofwords, 2015-10-28:
Hi Paul,

I only use 100% genuine Arduinos.

In my opinion, if this issue would be caused by a not 100% compatible USB chip, my Mac should simply not recognize and ignore it or transfer would return an error message.

On the contrary, when Arduino IDE tries to upload my sketch, it crashes and also makes my bluetooth mouse crash as well: only way to close it is kill it from shell or from “Cmd+Alt+Esc” (luckily touch pad keeps working). Afterward, my Mac fan starts slowly to increase its speed till maximum. At that point, I can only shut my Mac down (it takes ages to turn off, ten times longer than usual).

Best Regards.

Luca

wizardofwords, 2015-10-28:
Hi All,

Today I have bought a Yún (original, obviously!) and I have tried to upload a sketch: it does not work anyway, but IDE has not crashed and it has not crashed my bluetooth mouse.

Also it has displayed the following error message:

“Arduino:1.6.6 Hourly Build 2015/10/27 05:48 (Mac OS X), Scheda:"Arduino Yún"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/luca/Documents/Arduino/libraries" -fqbn=arduino:avr:yun -vid-pid=0X2341_0X8041 -ide-version=10606 -build-path "/var/folders/c3/pdvvh4zx4ydgyfv3l9643tt40000gn/T/build59155785b93a90ad5b7d283c25fce574.tmp" -warnings=null -prefs=build.warn_data_percentage=75 -verbose "/prg/C/Arduino/ShiftRegisterTest/ShiftRegisterTest.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/luca/Documents/Arduino/libraries" -fqbn=arduino:avr:yun -vid-pid=0X2341_0X8041 -ide-version=10606 -build-path "/var/folders/c3/pdvvh4zx4ydgyfv3l9643tt40000gn/T/build59155785b93a90ad5b7d283c25fce574.tmp" -warnings=null -prefs=build.warn_data_percentage=75 -verbose "/prg/C/Arduino/ShiftRegisterTest/ShiftRegisterTest.ino"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_YUN -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8041 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Yun"'  "/var/folders/c3/pdvvh4zx4ydgyfv3l9643tt40000gn/T/build59155785b93a90ad5b7d283c25fce574.tmp/sketch/ShiftRegisterTest.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_YUN -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8041 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Yun"' "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/yun" "-I/Users/luca/Documents/Arduino/libraries/ShiftRegister" "/var/folders/c3/pdvvh4zx4ydgyfv3l9643tt40000gn/T/build59155785b93a90ad5b7d283c25fce574.tmp/sketch/ShiftRegisterTest.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_YUN -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8041 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Yun"' "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/yun" "-I/Users/luca/Documents/Arduino/libraries/ShiftRegister" "/var/folders/c3/pdvvh4zx4ydgyfv3l9643tt40000gn/T/build59155785b93a90ad5b7d283c25fce574.tmp/sketch/ShiftRegisterTest.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_YUN -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8041 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Yun"' "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/yun" "-I/Users/luca/Documents/Arduino/libraries/ShiftRegister" "/var/folders/c3/pdvvh4zx4ydgyfv3l9643tt40000gn/T/build59155785b93a90ad5b7d283c25fce574.tmp/sketch/ShiftRegisterTest.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_YUN -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8041 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Yun"' "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/yun" "-I/Users/luca/Documents/Arduino/libraries/ShiftRegister" "/Users/luca/Documents/Arduino/libraries/ShiftRegister/ShiftRegister.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_YUN -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8041 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Yun"' "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Ardui [... cut ...]
         Using Port                    : /dev/cu.usbmodem1A21
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
avrdude: ser_open(): can't open device "/dev/cu.usbmodem1A21": No such file or directory
avrdude: ser_send(): write error: Bad file descriptor
Problema di caricamento sulla scheda. Guarda http://www.arduino.cc/en/Guide/Troubleshooting#upload per suggerimenti"

Unfortunately part of the messages are in Italian, language of my Arduino IDE.

I hope it helps.

Regards.

Luca

matthijskooijman, 2015-10-29:
Are you sure /dev/cu.usbmodem1A21 is the right port to use? A friend of mine was using an Arduino Uno with his OSX laptop yesterday, and had to use /dev/tty.something IIRC. Don't take my word for it though, I'm not a Mac user at all :-)

wizardofwords, 2015-10-29:
Hi Matthijs,
Honestly I would expect it was "/dev/tty.usbmodem1A21", but only "cu" ports are listed in Arduino IDE...
Regars.
Luca

Luca Moscardini

-------- Messaggio originale --------

Data: 29/10/2015  08:03  (GMT+01:00)

matthijskooijman, 2015-10-29:
That suggests to me that no serial port for the Arduino is available at all, and this cu port is just your normal serial port, or an internal one, or whatever. Does the cu port disappear when you unplug the Arduino?

wizardofwords, 2015-10-29:
Hi Matthijs,
When I connect Arduino to my Mac two devices are created:
/dev/cu.usbmodem1A21
and
/dev/tty.usbmodem1A21
but just "cu" one is listed in Arduino IDE.
When I remove Arduino, also both devices are removed.
Regards.

Luca Moscardini

-------- Messaggio originale --------

Data: 29/10/2015  10:38  (GMT+01:00)

matthijskooijman, 2015-10-29:
Right. I suspect the tty one is the correct one, but I'm not entirely sure. I do think something was fixed in the serial port listing recently. Are you using a recent hourly build, or a release build?

PaulStoffregen, 2015-10-29:
Normally the "cu" device is the proper one to use.

The "tty" device has additional support for the "RI" (Ring Indicator) handshake signal, present on old serial interface dialup modems.  If you open the device in blocking mode, the tty version will block until the phone rings.  This is a traditional (or "ancient") BSD Unix behavior, to create 2 devices for modems... one intended to dialing ("cu") and the other intended for answering ("tty").

The JSSC library (and also the older RXTX library) used by Arduino opens the device in non-blocking mode to work around this, so with the Arduino IDE there's no difference.  Old versions of Arduino show both.  Quite some time ago, code was added to only show the "cu" devices in the Tools > Port menu.  The fact you're not seeing the "tty" version is intentional, not a bug.

matthijskooijman, 2015-10-29:
@PaulStoffregen, thanks for clarifying, I learned something again :-)

So the real error is:

    avrdude: ser_open(): can't open device "/dev/cu.usbmodem1A21": No such file or directory

Somehow the device file does not exist (anymore) when avrdude wants to open it?

wizardofwords, 2015-10-29:
Hi Matthijs,
Yes, I use last nightly release and also serial port last release.
Regards.
Luca

-------- Messaggio originale --------

Data: 29/10/2015  12:44  (GMT+01:00)

wizardofwords, 2015-10-29:
Dear Paul,
In fact I did not considered its absence as a bug, but just an additional information for developers, hoping to help them to fix the real bug.
Thanks and Regards.
Luca Moscardini

-------- Messaggio originale --------

Data: 29/10/2015  12:54  (GMT+01:00)

PaulStoffregen, 2015-10-29:
Well, this is a mysterious problem.  I can tell you that doesn't happen on my Mac with El Capitan.

wizardofwords, 2015-10-29:
Hi Paul,

I definitively agree with you: it is a weird issue.

I am ready to let you have a look via TeamViewer: maybe you can see something macroscopic that I can’t.

Kind Regards.

Luca Moscardini

odbol, 2015-11-17:
So I have this same problem with my Arduino Due, and I've been doing a lot of testing to narrow it down. It only happens on El Capitan.

When I plug in the Due to a El Capitan Mac, I see this error in the Console system logs:

`11/16/15 9:37:19.000 PM kernel[0]: 049249.868982 IOUSBHostHIDDevice@14100000,2: IOUSBHostHIDDevice::interruptRetry: resetting device due to IO failures`

It repeats the error about every 5 seconds.

Could this be some kind of driver problem? An unsigned ktext disabled by SIP? Or a problem with the Atmel USB system? Should we submit a bug report to Apple?

I've made a test app to test the most basic failure case using JSSC: https://github.com/odbol/JSSCTest.git

Tested by making a runnable JAR from that project, and running it:

````
$ java -jar /Users/leiyumac/Downloads/Arduino-JSSC-Test2-delay.jar 

Running tests

Connecting

Nov 16, 2015 9:54:35 PM java.util.logging.LogManager$RootLogger log

SEVERE: Failed to open serial port. Port name - /dev/tty.usbmodem1411; Method name - openPort(); Exception type - Port not found.

Testing

Requesting update -1: BF7B01

Test failed: 

jssc.SerialPortException: Port name - /dev/tty.usbmodem1411; Method name - writeBytes(); Exception type - Port not opened.

    at jssc.SerialPort.checkPortOpened(SerialPort.java:878)

    at jssc.SerialPort.writeBytes(SerialPort.java:348)

    at com.odbol.jssctest.ArduinoJSSC.sendData(ArduinoJSSC.java:71)

    at com.odbol.jssctest.ArduinoJSSCTest.testSysexResponses(ArduinoJSSCTest.java:112)

    at com.odbol.jssctest.ArduinoJSSCTest.main(ArduinoJSSCTest.java:166)

    at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)

    at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)

    at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)

    at java.lang.reflect.Method.invoke(Method.java:483)

    at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)

Finished tests
````

odbol, 2015-11-17:
@cmaglie could you perhaps explain how you fixed this? It's not even working with the very basic JSSC test app I posted in the previous comment, and I'm having trouble understanding how your commit #4006 fixed it. Any advice would be much appreciated! Thanks!

cmaglie, 2015-11-17:
The fix in #4006 is in this line:
https://github.com/arduino/Arduino/pull/4006/files#diff-b5037ff11107550e3c087a34d6da5a69R76

basically the IDE now ignores the case when detecting serial port VIP/PID. This solve the issue about "disappearing" com port in ports list, but there should be more since some users, like you, are experiencing upload failures on OSX El Capitan.

> 11/16/15 9:37:19.000 PM kernel[0]: 049249.868982 IOUSBHostHIDDevice@14100000,2: IOUSBHostHIDDevice::interruptRetry: resetting device due to IO failures

Unfortunately this log doesn't tell me much, but it may be an hint to understand what the real cause is.

32u4, 2015-11-23:
I've got a pro micro with atmega 32u4. This used to work flawless, but now (even with latest build)
i can't upload, use serial port etc... Same as described in this topic. But it is even that bad that my mac completely stops working properly after trying to upload. Kernel_task process uses more than 100% cpu.
Can't shutdown or reboot anymore.

Even a clean install of osx (el capitan) and  arduino don't give me any progress.
Anny help would be appreciated.

odbol, 2015-11-24:
So I found a fix for my app, which was actually pretty easy. I just lengthened the delay between after scanning for USB devices (getPortNames() function in JSSC), and before trying to openPort(). Seemed to fix the problem:

````
     	String[] portNames = SerialPortList.getPortNames();
     	 
 		 System.out.println("Found " + portNames.length + " available ports.");
 		 
 		 for(int i = 0; i < portNames.length; i++){
 		     System.out.println(portNames[i]);
 		     
 		     result = portNames[i];
 		 }
 
 		// wait a bit since otherwise you get a "port open" exception if trying to open directly after calling this.

                // this is the magic part: increased from 100 to 900ms
		try { Thread.sleep(900); } catch (InterruptedException ie) {}
// this is the old one that gave a "port not found" exception
//		try { Thread.sleep(100); } catch (InterruptedException ie) {}

````

ffissore, 2015-11-24:
@odbol you should report your findings to jssc author https://github.com/scream3r/java-simple-serial-connector

mralexgray, 2015-12-30:
Why is this closed?  It is NOT fixed.

> Third party vendors must modify their drivers and applications in order to comply with the new policy.

OK, _third-party_.. Is this a case of "`Wont Fix`"?

matthijskooijman, 2015-12-30:
@mralexgray, I suspect that there might be multiple issues with serial ports on El Capitan. One of them, related to serial device enumeration was fixed, but I'm not sure if all of the comments in this issue are about them. If you still have a problem with the latest hourly build, consider opening a new issue (after searching the tracker for open issues matching your problem, if you didn't already).

PaulStoffregen, 2015-12-30:
FWIW, we've been seeing problems with Teensy when using USB Serial at high data rates with El Capitan.  Here's the conversation.

https://forum.pjrc.com/threads/31043-Teensyduino-1-26-Beta-1-Available?p=89113&viewfull=1#post89113

A couple weeks ago I put some serious effort into investigating the issue with a USB protocol analyzer.  I'm afraid the news is not good (at least for El Capitan).  All the data was being properly sent, with nothing unusual on the USB cable corresponding to the occasional data loss seen my programs running on El Capitan.  I tried adding code to occasionally send a ZLP within long sequences of full size packets as a possible workaround, but unfortunately that made no difference.

This rapid data flow problem may be completely unrelated to all the problems above.

lhcavalcantis, 2016-01-19:
I had the same problem with FTDIs, in the end I installed a correct driver and uploaded the program, but after using the FTDI for a few days in the USB port, the port don`t work anymore until I restart the Macbook. So, if I use any of the USB ports on Mac it stops working. Are there someone with that problem?

Breidenbach, 2016-01-19:
This is a persistent problem with El Capitan and the Arduino IDE.  Since upgrading, I have yet to successfully connect to my genuine Arduino Uno when using El Capitan even after trying about a half dozen "solutions" presented on the internet.  Fortunately I run Windows XP with VMware Fusion under El Capitan, and the IDE does connect in that environment!  But then, not everyone has that option.

lrothschild, 2016-01-25:
After upgrading to El Capitan on my 2014 MacBook Pro, my "Genuino" made-in-Italy Arduino Uno has acted erratic and finally stopped uploading.  The IDE has been unable to find any port except cu.Bluetooth-Incoming-Port, which of course does not work for uploading.  I've scoured the Arduino forums, but have not found any workable solution.  All suggestions welcome!

DrewOJensen, 2016-01-26:
I second Irothschild issues. I've got an ArduinoISP that is recognized by OS X system information, but in the IDE, I only get the Bluetooth-Incoming-Port. I also am on El Capitan.

Breidenbach, 2016-01-28:
Same problem, bluetooth gets lost, no connection to arduino using PR-4490-BUILD-495 (1.6.8)
This file is Apple complaining about wake-ups, but might be helpful otherwise.
After updating to latest El Capitan version - 10.11.13 - took a screen shot of the console.
If there is a better way to get info to help find the problem, tell me what to do.
[image]

squidlyboy, 2016-03-06:
Mac OS X EL Capitan 10.11.3 - Arduino 1.6.7
This is still broken.  Completely locks up my Mac when its uploading any sketch to the board via USB serial

32u4, 2016-03-08:
@squidlyboy : Did you try using a usb hub?
I had exactly the same problem as you have and i accidentally discovered
that all my boards (pro mini, pro micro, nano, mini) work as long as i use a usb hub. 
Without a hub my mac locks up... can't even shutdown properly.

Breidenbach, 2016-03-08:
When I use a hub, I get:
avrdude: ser_open(): can't open device "/dev/cu.usbmodem1A21": No such file or directory
ioctl(“TIOCMGET"): Inappropriate ioctl for device

When I don’t use a hub, I lose my Bluetooth mouse and keyboard, and the sketch is not uploaded.  Sometimes it resets the Arduino, but still doesn’t upload.  This is after moving the app to the root directory, then copying it to Applications.

BTW:  @Richard Loxley  Are you familiar with Apple’s new “sandbox” security?  It’s not just a matter of permissions anymore.
http://www.macworld.com/article/2948140/os-x/private-i-el-capitans-system-integrity-protection-will-shift-utilities-functions.html

And I have tried to disable this feature, but it didn’t help.

Hal

sandeepmistry, 2016-03-09:
@Breidenbach which board are you using in https://github.com/arduino/Arduino/issues/3928#issuecomment-175933160?

Also, are you using a powered or unpowered USB hub?

The ```Symantec Vulnerability Protection has found and blocked an attack (23387).``` entry in the console seems suspicious, does it occur every time you trying to upload a sketch?

Breidenbach, 2016-03-09:
> @Breidenbach <https://github.com/Breidenbach> which board are you using in #3928 (comment) <https://github.com/arduino/Arduino/issues/3928#issuecomment-175933160>?

Powered

Breidenbach, 2016-03-09:
Powered hub, Genuine Arduino Uno

bmikhael, 2016-04-09:
I've just downloaded the nightly build dated April 08 2016 on OS X El Capitan 10.11.3
I still face the same issue, I can't upload a sketch to the board.
Arduino IDE see the board connected to /dev/cu.usbmodem1421 but no luck uploading a sketch to the board.

DanceOfDragon, 2016-04-09:
Hi, @bmikhael  you might need to update your Arduino to the latest version as well if you haven't done so. Check your version of "board manager" etc. I am running the latest EI Capitan and a cu version UNO, it is working fine.

bmikhael, 2016-04-09:
@DanceOfDragon, I've the latest nightly build for Arduino IDE and Intel Curie Boards v1.0.5, that's the latest I can find, and it's not working!

cba76, 2016-05-04:
I'm not sure why this is closed as it's not resolved. I'm not able to see any ports except for the one bluetooth port that is incorrect. I'm running EC 10.11.4 with the latest daily build of Arduino at the time of this posting and can't upload anything to my ATmega 2560. Anyone have any ideas?

FoRTu, 2016-06-04:
I had the same problem and the only solution I found was connect a external power supply to the Arduino. I think that the USB power isn't enough to supply the Arduino and our circuits, so the board is detected sometimes and sometimes no, depending the power needed on our circuit.

So try to connect the Arduino board to a external powered USB Hub, or  a external power supply.

mosspassion, 2016-06-28:
I'm having this issue with a [Bare Conductive Touch Board](http://faqs.bareconductive.com/hc/en-gb/sections/201482111-Uploading-Code). I have a genuine Uno that mounts fine, and is selectable in the **Tools -> Port:** menu, but the Touch Board doesn't even show up when I check the command line `ls -ha /dev/tty.*; ls -ha /dev/cu.*`

Maybe @stefandz or @peter-krige could point me in the right direction?

I am using the Arduino IDE 1.6.9 on a Macbook Pro 9,1 in El Capitan 10.11.5.

stefandz, 2016-06-29:
@mosspassion it sounds like you have this solved for now - different USB ports will be able to supply different amounts of power, and external circuitry attached to the board will certainly draw more power, potentially exceeding this limit. 

The fact that changing the USB cable made a difference is not surprising either - due to the number of leads designed purely for charging mobile phones (i.e. without the data conductors in place) we find a number of people having this issue. Certainly something to consider if you ever design your own board.

If you have further issues with this, please send an email to info@bareconductive.com as this certainly isn't an issue for the Arduino development team - they have enough to do already!

tvervest, 2016-09-21:
I ran into the same issue today, so I figured I'd look into it and add any info I could find. When connecting the MKR1000 to my MacBook Pro (using El Capitan 10.11.6) the board boots, but it isn't listed in my devices (not even in the `/dev` directory). However, I do get the following messages in my syslog which might be of use:

```
21/9/16 18:16:47,000 kernel[0]: 000302.278333 IOUSBHostDevice@14100000: IOUSBHostDevice::getDescriptorGated: compliance violation: USB 2.0 9.3.5: device returned more than wLength data
21/9/16 18:16:47,000 kernel[0]: 000302.279216 Genuino MKR1000@14100000: IOUSBHostDevice::getDescriptorGated: compliance violation: USB 2.0 9.3.5: device returned more than wLength data
21/9/16 18:17:02,000 kernel[0]: 000317.453811 IOUSBHostDevice@14100000: IOUSBHostDevice::getDescriptorGated: compliance violation: USB 2.0 9.3.5: device returned more than wLength data
21/9/16 18:17:02,000 kernel[0]: 000317.454532 Genuino MKR1000@14100000: IOUSBHostDevice::getDescriptorGated: compliance violation: USB 2.0 9.3.5: device returned more than wLength data
```

I think this might be a lead to why the device doesn't mount, however I'm not sure what could cause this and what I could do about it - if anything at all. I'll investigate a bit more and report any useful findings here :)

facchinm, 2016-09-21:
Hi @tvervest ,
could you test the latest SAMD core? This looks like a problem with USB descriptors, more than an hardware problem. To upload a sketch, put the board in bootloader by double tapping the reset button

tvervest, 2016-09-21:
Hey @facchinm, I was just typing an update that the problem was resolved with a double-tap of the reset button xD

It seems to work fine now, not sure if it matters but I've also installed [the USB driver as described by @mosspassion](https://github.com/arduino/Arduino/issues/3928#issuecomment-229212128).

borch84, 2016-10-06:
Hello, I am trying to connect my huzzah esp8266 to my mac os el capitan 10.11.6, but I can only see these boards in my IDE
[image]

The IDE does not detect the serial port. I also checked in my file system but I don't see a device pointing to my huzzah.

I've checked my system.log for messages but I don't see any, not even kernel related messages, as stated by other users. 

I've changed the USB cables. I've tested the same ESP8266 in another Linux machine and it was recognised by the IDE. 

Previous users reported problems with Arduino Uno boards, I've haven't seen this issue with a Huzzah ESP 8266 board over the internet. 

The sketch gets compiled without error, but when I upload the sketch I got:
sptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
An error occurred while uploading the sketch
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.itead-DevB
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.itead-DevB at 115200
error: cannot access /dev/cu.itead-DevB

error: espcomm_open failed
error: espcomm_upload_mem failed

That is pointing to /dev/cu.itead-DevB, but this is not the real huzzah. 

Has anyone experienced the same issue? Any advices? Thank you.

[10 more replies]