arduino/Arduino #3929: digitalPinToInterrupt() does not exist

wiktor340, 2015-10-07:
Script from : https://www.arduino.cc/en/Reference/AttachInterrupt

Outcome: error: 'digitalPinToInterrupt' was not declared in this scope.

There is a lot of writing about how recommended it is but it does not work.

int pin = 13;
volatile int state = LOW;

void setup() {
    pinMode(pin, OUTPUT);
    attachInterrupt(digitalPinToInterrupt(pin), blink, CHANGE);
}

void loop() {
    digitalWrite(pin, state);
}

void blink() {
    state = !state;
}

harryboo, 2015-10-07:
@wiktor340:

This is a macro defined in 'pins_arduino.h', but not in all variants.
What variant (board) are you using?

agdl, 2015-10-07:
Furthermore, are you sure that you can attach an interrupt to that pin? Please look at the table of the pins you can use depending on the board you have!

wiktor340, 2015-10-07:
Hi, 
I  work with Mega 2560.

--
Wiktor
Permanent Dreamer

PaulStoffregen, 2015-10-07:
I just tried with the latest hourly build.

Two boards fail: Arduino Ethernet and Arduino Gemma

PaulStoffregen, 2015-10-07:
@wiktor340 Perhaps you should try the latest hourly build.  Arduino Mega 2560 does compile properly with your sketch.  You almost certainly have an old or unsupported version of the Arduino software.

agdl, 2015-10-22:
@wiktor340 I'm closing this for lack of feedback. Reopen it if needed

PaulStoffregen, 2015-10-22:
I'm reopening this issue.  The problem is **NOT** fixed.

To reproduce the problem, copy this code into Arduino, and select either "Arduino Ethernet" or "Arduino Gemma" from Tools > Boards.  Then click Verify.  You will quickly see those 2 boards do not have digitalPinToInterrupt() properly defined!

    int pin = 13;
    volatile int state = LOW;

    void setup() {
      pinMode(pin, OUTPUT);
      attachInterrupt(digitalPinToInterrupt(pin), blink, CHANGE);
    }

    void loop() {
      digitalWrite(pin, state);
    }

    void blink() {
      state = !state;
    }

Please do not close this issue until Arduino is able to verify this code with those boards selected.

agdl, 2015-10-23:
@PaulStoffregen please close this if everything looks good for you

PaulStoffregen, 2015-10-23:
Looks good.  Confirmed, both boards verify without error using the latest hourly build.  :)

closed 2015-10-23 by PaulStoffregen

NicoHood, 2015-10-23:
@wiktor340 you should download the latest IDE at arduino.cc (1.6.5) and not use an old one (probably 1.0.5)

Cant you make the ubuntu repos to update their version or at least remove it so people dont use the old 1.0.5 anymore?

PhilD66, 2017-01-09:
I am using an ATmega328 (piduino) attached to a Raspberry Pi 3. I have run 'sudo apt-get install arduino' on the pi to get the latest version and I still get this compile error.

This is the line that provokes the error (and interruptPin is set to 2)...

  attachInterrupt(digitalPinToInterrupt(interruptPin), increment, RISING);

Do I need to fetch the latest arduino install some other way?

facchinm, 2017-01-09:
Hi @PhilD66 ,
the version shipped with debian/raspbian is really old, you can get the latest version [here](https://www.arduino.cc/download_handler.php?f=/arduino-1.8.0-linuxarm.tar.xz)

PhilD66, 2017-01-09:
Thanks @facchinm , I have done as you suggested and I now have 1.8.0 installed. It does not have an option for me to select the 'Gertboard with ATmega328 (GPIO)' nor the Programmer 'Raspberry GPIO'.

facchinm, 2017-01-09:
I believe you followed this guide (https://projects.drogon.net/raspberry-pi/gertboard/arduino-ide-installation-isp/) . All stuff is very hardcoded and should be performed in a much more compliant way; you'd better contact the author and ask for a "Board Manager" like package (which should resemble the attached one)
[gertboard.zip](https://github.com/arduino/Arduino/files/693212/gertboard.zip)

PhilD66, 2017-01-09:
Thanks. OK, I understand that this setup is non-standard. I will check with Gordon as suggested. As a final parting question...I have seen elsewhere that 'digitalPinToInterrupt(pin)' can be replaced with a const or variable that represents the desired value. Is that correct and if so, what would I put for digital pin 2?

PaulStoffregen, 2017-01-09:
Looks like Gordon is only publishing support for ancient versions of Arduino.  This is just one of many features that Arduino has added.  You're likely to run into all sorts of other issues.  The fundamental problem is using very old Arduino software with newer code which was designed with and depends upon features in modern Arduino versions.

But if you want a quick hack that might solve your immediate issue, where the define for digitalPinToInterrupt used on most boards with the '328 chip.

https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/standard/pins_arduino.h#L79

Copy that into your code and maybe it will solve this compile error.  But understand you're trying to use code someone wrote and tested with a modern version of Arduino.  It was never probably tested with such an old version.  If you're lucky all the problems will be this easy, where you get errors *before* you upload.

PhilD66, 2017-01-09:
Thanks to @facchinm and @PaulStoffregen : I can get my code to compile but I acknowledge that there are likely to be other problems using an out-of-date and modified version of Arduino. I sense a level of disapproval here that leads me to believe I should be using an official Arduino board instead of the one I have now. Understood. Thanks anyway.

PaulStoffregen, 2017-01-09:
The problem isn't 3rd party hardware.  That's fine.  The issue here is publishing for version 1.0.5 from May 2013.  That's nearly **4 years old** software!