arduino/Arduino #3932: [Feature Request] Add IDE as board dependancy

NicoHood, 2015-10-07:
See this:
http://forum.arduino.cc/index.php?topic=349752

The workaround works but is not nice.

The same for libraries would also be nice.
If a library has the workaround added, the error is displayed, but with a whole bunch of other errors. Beginners might be confused and dont read the first error carefully. There was also a discussion (issue) to not display any further error after the first.

matthijskooijman, 2015-10-08:
> If a library has the workaround added, the error is displayed, but with a whole bunch of other errors. 

Doesn't #error halt further processing?

Having a "minimal version number" might be a useful addition, though.

ffissore, 2015-10-08:
Thanks for bringing this out. It was in the plans but never scheduled

per1234, 2015-10-08:
It would be nice if a version range could be specified e.g. 1.6.4-1.6.5, <=1.6.6, >=1.6.6, 1.6.5

matthijskooijman, 2015-10-08:
Might be useful to look at the version number specs that Debian uses.

Typically, you would just have just a minimal version. I can see two usecases for an upper limit:
 1. Some change in the IDE breaks a library / core (and fixing this in the library/core breaks compatibility with lower IDE versions, or just hasn't happened yet).
 2. Some specific IDE version breaks this library / core, due to an IDE bug that has again been fixed later. In this case, you need to specify multiple ranges, e.g. 1.6.0 - 1.6.2, 1.6.4+ for a bug in 1.6.3 that breaks

NicoHood, 2015-10-08:
Also there are different avr core with r5 for example. This should be added as well. (avr core dependency)
Also I wanted to create a board definition r2 (after I noticed my release needs some more minor tweak but I dont want to push the version number because its connected with the bootloader) but this is not in the #.#.# Number schema. But the IDE does the same. So this should probably supported as well.