arduino/Arduino #3935: Night build - gives warnings for Due "deprecated recipe.c.combine.pattern"

rogerclarkmelbourne, 2015-10-08:
Nightly 2-15/10/07  06:43

Installed Arduino SAM boards, 

Select the Due

Compile a blank sketch I get the following warnings 

'''
Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mcpu={build.mcu} "-T{build.variant.path}/{build.ldscript}" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" "-L{build.path}" -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "{build.path}/syscalls_sam3.c.o" {object_files} "{build.variant.path}/{build.variant_system_lib}" "{build.path}/{archive_file}" -Wl,--end-group -lm -gcc, automatically converted to recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mcpu={build.mcu} "-T{build.variant.path}/{build.ldscript}" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" "-L{build.path}" -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "{build.path}/core/syscalls_sam3.c.o" {object_files} "{build.variant.path}/{build.variant_system_lib}" "{build.path}/{archive_file}" -Wl,--end-group -lm -gcc. Consider upgrading this core.
Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'

Sketch uses 10,660 bytes (2%) of program storage space. Maximum is 524,288 bytes.
'''

ffissore, 2015-10-08:
Duplicate of #3821. See https://github.com/arduino/Arduino/issues/3821#issuecomment-141894840

closed 2015-10-08 by ffissore

rogerclarkmelbourne, 2015-10-08:
Sorry. I googled for the issue but it didnt return any results :-(

rogerclarkmelbourne, 2015-10-08:
Note. There are also 2 lines of warnings not in the duplicate bug

WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'

I'm not sure if you are addressing these at the same time

ffissore, 2015-10-08:
Also these have been fixed. But until these updates to SAM core won't get released, the warning will keep on popping up