arduino/Arduino #3937: tone() for Due

chrisrichards01, 2015-10-08:
Please add to documentation that tone() does not currently work with the Due.

VogonJeltz, 2015-10-08:
... and then please add a lib that would work with the Due...

q2dg, 2015-10-08:
...and what about Zero? I'm in Europe and I can't try it.

aethaniel, 2015-10-10:
@q2dg tone has been implemented for the Zero a while ago ;-)
https://github.com/arduino/ArduinoCore-samd/commits/master/cores/arduino/Tone.cpp

VogonJeltz, 2015-10-11:
but not for the Due (which is far longer on the market) ?

cmaglie, 2015-10-20:
Duplicate of #2362

closed 2015-10-20 by cmaglie

VogonJeltz, 2015-10-21:
I don't see how it now should work for the Due - can you provide a example source code please?

cmaglie, 2015-10-21:
The tone function is not implemented on the Due, and the code posted by @aethaniel is for the Zero board and cannot be used for the Due. That's why #2362 is still open.

VogonJeltz, 2015-10-21:
ok, thank you, I didn't understand this.
is there a timeline about when it will be available?

VogonJeltz, 2015-12-06:
https://github.com/arduino/Arduino/issues/2362

why hadn't this been fixed yet?!?