arduino/Arduino #3941: Editor and Auto Format have conflicting multi-line comment styles

per1234, 2015-10-09:
Using Arduino IDE 1.6.6 2015/10/08 05:43
- type: /*
- press enter key

editor produces the following:
```
/*
 * 
 */
```
- Tools>Auto Format

auto format changes it to:
```
/*

*/
```

Chris--A, 2015-10-09:
Just my 2 cents.

I do not use auto format, however the auto addition of `*` is quite annoying. I can understand auto bracket matching but this is a feature that doesn't really add to *ease of use*.

It'd be good to change the editor to reflect the auto-format style, rather than the other way around.

per1234, 2015-10-09:
>auto addition of * is quite annoying.

100% agree.
>I can understand auto bracket matching

I can't stand it. This feature may be useful to some but I really wish it was optional. My productivity using the Arduino IDE has dropped significantly since the new editor was added.
>It'd be good to change the editor to reflect the auto-format style, rather than the other way around.

Agree, I'd probably have to switch to using an external editor if Auto Format started adding * to every line of a multi-line comment. However, I think it more likely the fix will be for Auto Format to just not remove the * from each line though I'd prefer the editor's behavior to be changed instead.

Chris--A, 2015-10-09:
> I can't stand it. This feature may be useful to some but I really wish it was optional. My productivity using the Arduino IDE has dropped significantly since the new editor was added.

I'm with you on this, I hate it, however it seems to be what they wan't and I can see where it may possibly save a newbie. But still, it doesn't really do its job properly, anything other than creating a nested scope or a function still requires intervention by the user (classes, namespaces): https://github.com/arduino/Arduino/issues/3390

ffissore, 2015-10-28:
See https://github.com/bobbylight/RSyntaxTextArea/issues/151