arduino/Arduino #3944: AM/PM handling with 12-hour RTC setting?

damellis, 2015-10-09:
If you put the RTC library in 12-hour mode, how does it handle AM and PM? Is there any way to specify whether or not an hour is AM or PM? Or tell, when you get the time, whether it's AM or PM? If not, should we just eliminate 12-hour mode?

VogonJeltz, 2015-10-09:
maybe poll the time in 24 hour intermediately?
char * hourfmt= (hour24 != hour12 ? "PM" : "AM") ;

agdl, 2015-10-19:
@damellis you are right, at the moment, we have no method to understand the time representation, however it ca be simply implemented pollig the control bit RTC_MODE2_CTRL_CLKREPin RTC->MODE2.CTRL.reg (that you set with the rtc.begin(bool timeRep). I also wnt to add methods that returns to you a string for time and date. What do you think about these improvements?

damellis, 2015-10-19:
I think you could start by just dropping 12 hour support, and letting the
user's sketch do the translation from 24 hour to 12 hour time if they want
to print it that way. If we do add date formatting code, it would be good
to include 12 hour options there, but that also feels like an extension
that isn't necessarily a solution to this particular issue.

agdl, 2015-10-21:
@damellis @VogonJeltz I removed the 12 hour support as you suggested. It can be implemented in the sketch with no so may effort

https://github.com/arduino-libraries/RTCZero/commit/c7d65d21feb6bb60c69a277e73076f023f1b7ac7

agdl, 2015-10-28:
I'm closing this. Reopen it if there are further requests please

agdl, 2015-11-17:
We have this PR. What do you think about it?

agdl, 2015-11-17:
Adding @cmaglie and @facchinm to the loop

damellis, 2015-11-17:
Personally, I don't really see the value of 12 hour time. It just seems
more complicated to call one function to get the hours and another function
to get AM vs. PM, when you could just get one number from 0 to 23. Are
there people that are asking for 12 hour functions? What's the use case?

agdl, 2015-11-18:
@damellis ok i asked also to Gabriel his opinion, and he agreed to remove it. So I'm closing this again!

closed 2015-11-18 by agdl

ShoneVj, 2017-07-05:
i think its better to setDS1307 in 24 hr mode and later convert it to 12 hr .Bcoz if we config it in 12 hr mode it is not possible to read whether it is am or pm.. i not pretty sure about this any way.