arduino/Arduino #3950: IDE does not launch

Avamander, 2015-10-10:
IDE Version:   9 October 2015 13:15:11 GMT Nightly
OS: Ubuntu 15.04
Issue: IDE does not launch, throws an error (but latest stable build runs fine).

    :~/Arduino IDE$ ./arduino
    Picked up JAVA_TOOL_OPTIONS:
    Error: Could not find or load main class IDE.lib.splash.png

matthijskooijman, 2015-10-13:
I just tried the nightly build (12 October 2015 17:15:7 GMT, linux64) on Debian Jessie, which works as expected.

The error doesn't make any sense to me either. The main class listed looks like part of the splash screen passed to the the java -splash option, which shouldn't be used as the main class.

Could you run this command, in the directory where you extracted the nightly?

    /usr/bin/env bash -x arduino

This runs the script while logging all commands run. If you paste that output here, we might be able to tell what's going on.

Avamander, 2015-10-13:
~/Arduino IDE$ /usr/bin/env bash -x arduino
+++ readlink -f -- arduino
++ dirname -- '/home/a/Arduino IDE/arduino'
+ APPDIR='/home/a/Arduino IDE'
+ for LIB in '$APPDIR/java/lib/rt.jar' '$APPDIR/java/lib/tools.jar' '$APPDIR/lib/*.jar'
+ CLASSPATH=:/home/a/Arduino
+ for LIB in '$APPDIR/java/lib/rt.jar' '$APPDIR/java/lib/tools.jar' '$APPDIR/lib/*.jar'
+ CLASSPATH=:/home/a/Arduino:IDE/java/lib/rt.jar
+ for LIB in '$APPDIR/java/lib/rt.jar' '$APPDIR/java/lib/tools.jar' '$APPDIR/lib/*.jar'
+ CLASSPATH=:/home/a/Arduino:IDE/java/lib/rt.jar:/home/a/Arduino
+ for LIB in '$APPDIR/java/lib/rt.jar' '$APPDIR/java/lib/tools.jar' '$APPDIR/lib/*.jar'
+ CLASSPATH=:/home/a/Arduino:IDE/java/lib/rt.jar:/home/a/Arduino:IDE/java/lib/tools.jar
+ for LIB in '$APPDIR/java/lib/rt.jar' '$APPDIR/java/lib/tools.jar' '$APPDIR/lib/*.jar'
+ CLASSPATH=:/home/a/Arduino:IDE/java/lib/rt.jar:/home/a/Arduino:IDE/java/lib/tools.jar:/home/a/Arduino
+ for LIB in '$APPDIR/java/lib/rt.jar' '$APPDIR/java/lib/tools.jar' '$APPDIR/lib/*.jar'
+ CLASSPATH=':/home/a/Arduino:IDE/java/lib/rt.jar:/home/a/Arduino:IDE/java/lib/tools.jar:/home/a/Arduino:IDE/lib/*.jar'
+ export CLASSPATH
+ LD_LIBRARY_PATH='/home/a/Arduino IDE/lib'
+ export LD_LIBRARY_PATH
+ export 'PATH=/home/a/Arduino IDE/java/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games'
+ PATH='/home/a/Arduino IDE/java/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games'
+ [[ '' == *\-\-\u\p\l\o\a\d* ]]
+ [[ '' == *\-\-\v\e\r\i\f\y* ]]
+ [[ '' == *\-\-\g\e\t\-\p\r\e\f* ]]
+ [[ '' == *\-\-\i\n\s\t\a\l\l\-\b\o\a\r\d* ]]
+ [[ '' == *\-\-\i\n\s\t\a\l\l\-\l\i\b\r\a\r\y* ]]
+ SPLASH='-splash:/home/a/Arduino IDE/lib/splash.png'
++ echo
++ sed 's|-javaagent:/usr/share/java/jayatanaag.jar||g'
+ export JAVA_TOOL_OPTIONS=
+ JAVA_TOOL_OPTIONS=
+ JAVA=java
+ '[' -x /home/a/Arduino IDE/java/bin/java ']'
arduino: line 29: [: /home/a/Arduino: binary operator expected
+ java '-DAPP_DIR=/home/a/Arduino IDE' -Dswing.defaultlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel -splash:/home/a/Arduino IDE/lib/splash.png processing.app.Base
Picked up JAVA_TOOL_OPTIONS: 
Error: Could not find or load main class IDE.lib.splash.png

matthijskooijman, 2015-10-13:
Ah, the problem is the space in "Arduino IDE" I believe. Should be easy to fix, I'll create a PR in a minute.

PR #3973 (2015-10-14): Allow spaces in IDE install path on Linux
The startup bash script lacked quotes in some places, causing it to
interpret part of the path to the splash image as the main class and
fail to start:

arduino: line 29: [: /home/a/Arduino: binary operator expected
Error: Could not find or load main class IDE.lib.splash.png

For the -splash option, simply adding quotes was not sufficient. If no
splash screen was to be used (so when $SPLASH was empty), using
"$SPLASH" would result in an empty argument being passed to java, which
was then interpreted by java as the name of the base class.

To allow spac [... cut]
commit bef7bb5b (2015-10-14): Allow spaces in IDE install path on Linux

The startup bash script lacked quotes in some places, causing it to
interpret part of the path to the splash image as the main class and
fail to start:

arduino: line 29: [: /home/a/Arduino: binary operator expected
Error: Could not find or load main class IDE.lib.splash.png

For the -splash option, simply adding quotes was not sufficient. If no
splash screen was to be used (so when $SPLASH was empty), using
"$SPLASH" would result in an empty argument being passed to java, which
was then interpreted by java as the name of the base class.

To allow spa [... cut]
closed 2015-10-14 by cmaglie