arduino/Arduino #3951: Splash screen divided with dual monitors

nathansizemore, 2015-10-10:
When opening the IDE on Linux with dual monitors, the splash windows is centered between the two monitors instead of the center of any one monitor

~~~
Distributor ID:	Ubuntu
Description:	Ubuntu 15.04
Release:	15.04
Codename:	vivid
~~~

NicoHood, 2015-10-10:
I have the same issue, but never noticed it as bug.
Another more important bug is, that the graphical interface totally glitches when a screen is (un)plugged.

nathansizemore, 2015-10-10:
I'm using 2 16:9s.

I doubt it's the intended placement behavior, which is why I reported it as a bug.

matthijskooijman, 2015-10-13:
> Another more important bug is, that the graphical interface totally glitches when a screen is (un)plugged.

@NicoHood, please try to remain on-topic with your comments. If you think this is another important bug - report it as such. We can't properly track down bugs that are mentioned in off-topic comments, and they cloud the original report.

@nathansizemore Agreed that this is a bug, though I suspect that fixing it might need more work than the Arduino team can spare for such a minor issue. If you, or anyone else, can figure this out and submit a pullrequest to fix this, that would be nice. I'll leave it up to the devs to decide if they want to have a look at this, or just leave the bug.

nathansizemore, 2015-10-13:
@matthijskooijman Understandable. If I get enough free time in the coming weeks, I'll get it taken care of and send a PR

ffissore, 2015-10-28:
It's a known issue. Its source is the use of java `-splash` commad line param, which is buggy and unmaintained since the introduction of JavaFX. Proposed solutions are making your own splashscreen implementation. See http://stackoverflow.com/questions/1968880/center-java-splash-screen-on-multi-monitor

ijash, 2019-02-02:
is this still a thing? i still got the problem on linux mint 19

facchinm, 2019-02-04:
> is this still a thing? i still got the problem on linux mint 19

It is, any pull request fixing it will be merged immediately :smile:

pik10, 2021-11-12:
I have Fedora 35.  Arduino IDE installed from Flathub.  The splash screen is still divided between two monitors.