arduino/Arduino #3955: arm-none-eabi-gcc: internal compiler error: Segmentation fault (program cc1)

nplanel, 2015-10-11:
On ARCH Linux distribution (uptodate)
Fresh Arduino 1.6.5 + Arduino Due package

Crash append during WInterrupts compilation
```
Starting program: /home/nplanel/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -Dprintf=iprintf -MMD -mcpu=cortex-m3 -DF_CPU=84000000L -DARDUINO=10605 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER=Unknown -DUSB_PRODUCT=Arduino\ Due -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/system/libsam -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/system/CMSIS/CMSIS/Include/ -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/system/CMSIS/Device/ATMEL/ -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/cores/arduino -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/variants/arduino_due_x /home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/cores/arduino/WInterrupts.c -o /tmp/build5834105100023596365.tmp/WInterrupts.c.o
```

```
Using library SPI in folder: /home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/libraries/SPI 

/home/nplanel/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD -mcpu=cortex-m3 -DF_CPU=84000000L -DARDUINO=10605 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Due" -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/system/libsam -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/system/CMSIS/CMSIS/Include/ -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/system/CMSIS/Device/ATMEL/ -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/cores/arduino -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/variants/arduino_due_x -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/libraries/SPI /tmp/build5834105100023596365.tmp/wm8734.cpp -o /tmp/build5834105100023596365.tmp/wm8734.cpp.o 
Using previously compiled file: /tmp/build5834105100023596365.tmp/SPI/SPI.cpp.o
Using previously compiled file: /tmp/build5834105100023596365.tmp/variant.cpp.o
Using previously compiled file: /tmp/build5834105100023596365.tmp/dtostrf.c.o
/home/nplanel/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -Dprintf=iprintf -MMD -mcpu=cortex-m3 -DF_CPU=84000000L -DARDUINO=10605 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Due" -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/system/libsam -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/system/CMSIS/CMSIS/Include/ -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/system/CMSIS/Device/ATMEL/ -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/cores/arduino -I/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/variants/arduino_due_x /home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/cores/arduino/WInterrupts.c -o /tmp/build5834105100023596365.tmp/WInterrupts.c.o 
/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/cores/arduino/WInterrupts.c: In function 'attachInterrupt':
/home/nplanel/.arduino15/packages/arduino/hardware/sam/1.6.4/cores/arduino/WInterrupts.c:120:1: internal compiler error: Segmentation fault
 }
 ^
arm-none-eabi-gcc: internal compiler error: Segmentation fault (program cc1)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
/home/nplanel/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc returned 4
Error compiling.
```

nplanel, 2015-10-11:
source code
```
#include <SPI.h>

//enum WM8734_reg {
int  LINE_L = 0x00;
int  LINE_R = 0x01;
int  APATH = 0x04;
int  DPATH = 0x05;
int  POWERD = 0x06;
int  INTERF = 0x07;
int  SAMPL = 0x08;
int  ACTIVE = 0x09;
//};

const int slaveSelectPin = 10;

void setup() {
  // set the slaveSelectPin as an output:
  pinMode (slaveSelectPin, OUTPUT);
  digitalWrite(slaveSelectPin, HIGH);
  SPI.begin();
}

void setup_codec() {
  const int zerodB = 0x17;
  codecWrite(LINE_L, 0x00 || zerodB);
  codecWrite(LINE_R, 0x00 || zerodB);

  codecWrite(APATH, (1<<4)); // DAC select
  codecWrite(DPATH, 0x00);

  codecWrite(POWERD, 0x00 || (1<<1) || (1<<5) || (1<<6));
  codecWrite(INTERF, 0x00); // MSB right 16 bits
  const int SR = 0x04; // 44.1 KHz
  codecWrite(SAMPL, SR << 2);
  codecWrite(ACTIVE, 0);
}

void loop() {
  for (;;) {
    setup_codec();
    delay(3000);
  }

}

void codecWrite(int address, int value) {
  SPI.transfer(address);
  SPI.transfer(value);
  digitalWrite(slaveSelectPin, LOW);
  delay(1);
  digitalWrite(slaveSelectPin, HIGH);
}
```

facchinm, 2015-10-12:
Hi @nplanel, very strange problem (and no-one ever reported anything similar).
I've just tested with hourly IDE and SAM core from `master` (it should not solve the problem since the compiler and WInterrupt file have not changed in the meantime) on an updated Arch and everything is ok.
Wondering what can trigger a compiler segfault... Can you test using hourly IDE? Thanks!

nplanel, 2015-10-12:
I fixed the issue by removing (renaming) .arduino15 and redownload/reinstall the board

Aftermath I done a diff between the old and the new directory ; 
And I found some weird diff in sam3xa/svd/SAM3X8H.svd (like corruption)

-              4bitOffset>29</bitOffset>
+              <bitOffset>29</bitOffset>

and diff report me that the archive files differ .arduino15.old/staging/packages/sam-1.6.4.tar.bz2 and .arduino15/staging/packages/sam-1.6.4.tar.bz2 

Did you do a check on downloaded files ?

facchinm, 2015-10-12:
Yep, we always check that the downloaded files have the right size and sha256 hash (and I'm sure this check works, because developing cores I usually forget to update one and the IDE complains loudly :smile: ).
So, unless you found the first known sha256 collision (which would be a great discover, btw) I suppose the file got corrupted between the check and the extraction...
I'm assigning this to @ffissore to see what we can do, so thanks for pointing it out and for the accurate description!

nplanel, 2015-10-15:
hmm, further inspection show me that today the files are identical .... diff/sha256/md5 .... My computer didn't reboot (suspend to ram only)

So this bug look likes VFS issue it could be a RAM corruption "rowhammer bug" (no ECC, so possible) or an XFS bug or soft raid5 bug (I doubt).

Testing ram with memtester didn't found any issue.

So basically you can close this issue, it's look like it's on my side

cmaglie, 2015-10-15:
Ok, thanks for the update!

closed 2015-10-15 by cmaglie