arduino/Arduino #3956: Serial.write hangs during 1 second on arduino / CH340

pcorgier, 2015-10-11:
Every time a write is issued (not depending on sent size), it returns after 1 second.
Tested with 9600, 19200, 115200 bauds.

My arduino nano has a CH340 chip onboard, and I guess there is a timeout that systematicaly occures.

No problem with readString.

NicoHood, 2015-10-11:
That is impossible since the usb chip works independent from the MCU.
What do you mean "it returns" after a second? A specific sketch to test would be nice.

Also the CH340 is on no official board, so you wont get much help here. Buy official boards and support the people who answer your issues (nah, not me).

pcorgier, 2015-10-11:
I just checked my test case : in fact, write+flush operation lasts 436 microseconds (for 5 chars).
My remote program imediately sends a byte as an answer.
Total elapsed time between sending and acknowledge receive is 1 second.

pcorgier, 2015-10-11:
I just have added a Serial.setTimeout(20) call.
Elapsed time fall to 20ms.
So I am not facing to a bug, but I do not use the apropriate way for reading.

NicoHood, 2015-10-11:
Can you please specify this? Post a testsketch so we can try to reproduce this error?

pcorgier, 2015-10-11:
Here is my arduino sketch, correctly coded this time.

The remote aplication continuously reads the USB/serial COM and sends back a "o\n" string.
Measured elapsed time now never equals the timeout, but is in the range 10ms-13ms, that seems huge to me.
If I use HC-06 bluetooth instead of USB, elapsed time is between 30ms and 70ms !!!

But after all, I don't think that arduino hardware and software is concerned by this bad perfomances.

void setup()
{
Serial.begin(115200);
//Serial.setTimeout(20);
}

void loop()
{
  unsigned long send_date;
  unsigned long sent_date;
  unsigned long date;
  char buffer[20];
  byte n;

  while( 1 )
    {
    send_date = micros();
    Serial.write("#n10\n");
    Serial.flush();
    sent_date = micros();

    n = Serial.readBytesUntil('\n', buffer, sizeof(buffer));
    date = micros();

    buffer[n] = 0;
    Serial.print("send duration=");Serial.println(sent_date-send_date);
    Serial.print("elapsed time=");  Serial.println(date-send_date);
    Serial.print("n=");  Serial.print(n);
    Serial.print(" buffer=<");  Serial.print(buffer); Serial.println(">");

    delay(1000);
    }
}

NicoHood, 2015-10-11:
In your first post you talked about 1s not 10 ms? What about that?

However this is an OS delay. Depending how fast your PC is, the driver, the program, USB2 or3, HUB or direct port access 10ms is totally normal. also CH340 are no official boards.

pcorgier, 2015-10-11:
1s is the default reading timeout ; because my code was erroneous, timeout occured systematically.
When set to 10ms, timeout ouccured also.
Now my code is fixed ==> there is no more timeout, but performance are dramatically low for the use I planned for (realtime music instrument with remotely played notes) and I am affraid having to give up in the remote playing way :-(
Because for a musician, 5 ms is something like an eternity ...

closed 2015-10-11 by pcorgier