arduino/Arduino #3964: [DOC] Example code in https://www.arduino.cc/en/Tutorial/ArduinoZeroPowerConsumption doesn't implements any explicit thing about low power

q2dg, 2015-10-12:
Explanation seems good, but example code makes a "standard" use of analogRead() without any line showing described low power capabilities... Maybe the low power behaviour is automatic but I don't think so.

agdl, 2015-10-19:
@q2dg yes, it speak about the low power features that you have, but doesn't use any specific code instruction, because at the moment there is no Arduino code to implement these functions. The example shows the analogRead example because it is a way to show that just the use of a buck converter instead of a standard LDO running the same code decreased the consumed power.

q2dg, 2015-10-19:
Oook, perfectly understood!
Well...I'll close this issue, then (if you don't mind). When Arduino has code to implement these functions, we'll talk again.
Thanks!

q2dg, 2015-11-24:
Well...it seems "the others" have already done it: https://github.com/arduino-org/Arduino/tree/ide-org-1.6.1.x/libraries/Rest

agdl, 2015-11-27:
@q2dg I implemented this for now!

http://edit.arduino.cc/en/Tutorial/SleepRTCAlarm

agdl, 2015-11-27:
@q2dg I implemented this for now!

http://arduino.cc/en/Tutorial/SleepRTCAlarm

q2dg, 2015-11-28:
Ok, thanks. Few considerations, though: 

-"uint8_t" data type is not "natural" in Arduino world...maybe it would be worthy change it to "byte". Other Zero's example codes have the same issue

*In https://www.arduino.cc/en/Tutorial/ArduinoZeroPowerConsumption it is said: *"The device has two different software selectable sleep modes: idle & standby. In idle mode the CPU is stopped while all the other functions can be kept running. In standby all clocks and functions are stopped except those selected to continue running. "* So I infer that rtc.standbyMode() only manages standby mode, but idle doesn't...it isn't specified nowhere 

*I don't understand the code you have put in https://www.arduino.cc/en/Tutorial/SleepRTCAlarm . Why rtc.attachInterrumpt() calls a void function??? Shouldn't it be the loop() function that void function, (à la  https://www.arduino.cc/en/Tutorial/SimpleRTCAlarm) so only is necessary adding to that example the line rtc.standbymode() at the end of setup() and nothing more??

Thanks a lot for your patience

agdl, 2015-12-11:
@q2dg the interrupt is void because I don't want to do nothing in particular in that case, but I need an interrupt to wake up the zero from the standby mode. Maybe it is better to move the 
```
Serial.println("Awake!");
```
in the interrupt function. What do you think about it?

q2dg, 2015-12-14:
Well, my question is: in https://www.arduino.cc/en/Tutorial/SleepRTCAlarm sample sketch, why don't cut loop() code lines  to paste them into alarmMatch() body? (as it is done in https://www.arduino.cc/en/Tutorial/SimpleRTCAlarm sample sketch). That would result in a example more easy to understand (if I'm not wrong about it).

On the other hand, I insist:

-uint8_t data type is not "natural" in Arduino world...maybe it would be worthy change it to byte data type. Other Zero's example codes have the same issue

-Is it true that rtc.standbyMode() method only manages "standby mode" not "idle mode"?

Thanks!

agdl, 2015-12-14:
I don't want to cut and paste those line in the alarmMatch bosy because in this way it is clear that the code in the loop it is not executed till something happen. If you put it in the function you cannot understand that the board is stopped. For uint8_t data type you are right, I can easily change it. Yes the standbyMode deals only with standby and not to idle mode

q2dg, 2015-12-14:
Well, it is another way of understand it...ok, then.
I'll close the issue waiting for the change of uint8_t data type.
Thanks!!

closed 2015-12-14 by q2dg

agdl, 2015-12-14:
@q2dg datatype changed!