arduino/Arduino #3965: Enum cannot contain true/false

cb88, 2015-10-12:
But the following in a fresh sketch to break it.
```
enum ternary {
  unknown,
  true,
  false
};
```

NicoHood, 2015-10-12:
Those names are reserved. Thats all. Thats c/c++ i'd say.
You could use TRUE and FALSE

cb88, 2015-10-12:
Hmm... reserved to mean 0 and 1 or some other value? Which would mean you couldn't do true = someValue in an enum.

I mean having true and false in an enum makes sense... unless there is just some arbitrary limitation to not be able to do so. The wiki indicates this as doable here: http://playground.arduino.cc/Code/Enum

NicoHood, 2015-10-12:
true = true????
This makes no sense. however its a c++ question, so you might ask on stackoverflow for help. This is not an Arduino issue.

NicoHood, 2015-10-12:
Oh, the wiki aka playground is not official. Its.. yeah a wiki... with a crapy article. Someone should remove it.

cmaglie, 2015-10-12:
`true` and `false` are reserved keywords in C/C++, like `for`, `if`, `while`. This is a complete list:
http://it.cppreference.com/w/cpp/keyword

Simply you cannot use them for user-defined types.

And yes, the article on the playground should be fixed/removed...

closed 2015-10-12 by cmaglie