arduino/Arduino #3972: unable to find a register to spill in class 'POINTER_REGS'

danlu01, 2015-10-13:
I upgraded from Arduino IDE 1.0.6 to 1.6.5 and now my sketch have compile error:

unable to find a register to spill in class 'POINTER_REGS'

I've narrow it down to a function (I've modified it to try to understand the compile error).  Odd thing is if I remove any of the 4 line of code in the for loop, then it compiles ok.  Here's the test sketch with this issue:

------------------------
float dhistory[10];
float test;

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
  
  test = getSlope(dhistory);
}

float getSlope(float history[]) {
  float sumx = 0;
  float sumy = 0;
  float sumxy = 0;
  float sumxsq = 0;
  float rate = 0;
  int n = 10;
  
  for (int i=1; i< 11; i++) {
    sumx = sumx + i;
    sumy = sumy + history[i-1];
    sumy = sumy + history[i-1];
    sumxsq = sumxsq + (i*i);
  } 

  rate = sumy+sumx+sumxsq;
  return rate;
}
--------------------------------

NicoHood, 2015-10-13:
Please put your code into "```" brackets so it can be read better.
The simple solution to this is: Just DONT us floats on AVR. It just makes no sense at all. also it is important to know what board you used.

danlu01, 2015-10-13:
I'm compiling for Mega2560 but get same error if compile for UNO as well.  To me it's a compiler optimization error with pointers.

facchinm, 2015-10-13:
Yep, it's a gcc bug (https://gcc.gnu.org/bugzilla/show_bug.cgi?id=60040).
If you feel brave you could test if gcc 5.1.0 solves it following these instructions (https://github.com/arduino/Arduino/issues/660#issuecomment-120433193)

matthijskooijman, 2015-10-13:
I just tried 5.1, still broken. I've reduced this testcase to work outside of Arduino and added it to the gcc bug report.

@danlu01, it seems that enabling O2 (more optimizations) prevents this issue from occurring with your code. As a workaround, you could add this near the top of your sketch, to compile just that one function with O2 optimizations:

    float getSlope(float history[]) __attribute__((__optimize__("O2")));

(just tried, works for me)

danlu01, 2015-10-13:
Thanks for the workaround.. run into another compiler bug.  I'll create another ticket.

dgholstein, 2015-12-01:
All;

I have the same problem and it doesn't involve floats (though I use floats elsewhere).

As I uncomment more of my class, the constructor, *AD_Synth::AD_Synth(ThreeWire& com_port)*, I get the "register to spill" message.  I tried the **\__attribute\__** workaround with no success.

BTW:  When I wrote it all as C-based (not a class) code, the large struct worked just peachy, with essentially the same function as the method.

Sorry about the length of the code and especially the attributes -- but the AD4351 synthesizer is a complicated device.

Regards,
Dan
[synth.ino.txt](https://github.com/arduino/Arduino/files/48635/synth.ino.txt)
[AD_synth.cpp.txt](https://github.com/arduino/Arduino/files/48628/AD_synth.cpp.txt)
[AD_Synth.h.txt](https://github.com/arduino/Arduino/files/48630/AD_Synth.h.txt)
[ThreeWire.cpp.txt](https://github.com/arduino/Arduino/files/48632/ThreeWire.cpp.txt)
[ThreeWire.h.txt](https://github.com/arduino/Arduino/files/48631/ThreeWire.h.txt)

islamelhadi, 2016-10-07:
I was having the same issue and I simply resolved it. All what I did is downloading the LAST UPDATE (4 October 2016 14:52:59 GMT) for Arduino software in the software download page (Arduino software hourly builds) at the bottom left of the page. The sketch was successfully compiled.
HOPE this helps !
https://www.arduino.cc/en/Main/Software
[image]

islamelhadi, 2016-10-07:
I was having the same issue and I simply resolved it. All what I did is downloading the LAST UPDATE (4 October 2016 14:52:59 GMT) for Arduino software in the software download page (Arduino software hourly builds) at the bottom left of the page. The sketch was successfully compiled.
HOPE this helps !
[https://www.arduino.cc/en/Main/Software](url)
[image]

njh, 2016-10-26:
I just tried an hourly build "21 October 2016 16:13:52 GMT" - which came with 'avr-gcc (GCC) 4.9.2' but it didn't solve the problem for me.

    error: unable to find a register to spill in class 'POINTER_REGS'

vitasam, 2016-10-29:
Looks like I just got the same problem with Arduino 1.6.11. 
This line compiles Ok in multiple place of sketch:
`sprintf(msg_payload, "%s is ON", module_config.client_name);`

but in one place of the sketch this line:
`      sprintf(msg_payload, "%s connected", module_config.client_name);` 
returns an error:
```
...\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.14\cores\arduino\main.cpp:51:1: error: unable to find a register to spill in class 'NO_REGS'
...
        (subreg:QI (reg/f:HI 458) 1)) ...\GITprojects\Sourceforge\home-controller-code\Arduino\AERCM\AERCM.ino:185 1 {pushqi1}

     (expr_list:REG_ARGS_SIZE (const_int 1 [0x1])

        (nil)))
```

With just one parameter a compilation is Ok:
`sprintf(msg_payload, "%s", module_config.client_name);`

matthijskooijman, 2016-10-30:
According to the GCC bug, this will be fixed in GCC 7.0. They might backport the fix to earlier versions, if so I think it should be mentioned in the bug report as well. https://gcc.gnu.org/bugzilla/show_bug.cgi?id=60040

ZeusDataBinder, 2018-07-09:
Hello All,
I am still getting the PONITER_REGS error in the UNO board. I am using 1.8.5 IDE!

reivaxy, 2018-07-13:
So am I.