arduino/Arduino #3974: Compiler issue:  must be const in order to be put into read-only section by means of '__attribute__((progmem))'

danlu01, 2015-10-13:
Just upgraded from v.1.0.6 to 1.6.5 and run into another compiler issue.  Compiled fine with 1.0.6 but failed with 1.6.5.

In file included from ...libraries\Time\DateStrings.cpp:11:0:
...\Time\DateStrings.cpp:41:22: error: variable 'monthNames_P' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
 PGM_P monthNames_P[] PROGMEM = 

PGM_P is define as "const char *"

Here's the code:
/* DateStrings.cpp
 * Definitions for date strings for use with the Time library
 *
 * No memory is consumed in the sketch if your code does not call any of the string methods
 * You can change the text of the strings, make sure the short strings are each exactly 3 characters 
 * the long strings can be any length up to the constant dt_MAX_STRING_LEN defined in Time.h
 * 
 */

#if defined(__AVR__)
#include <avr/pgmspace.h>
#else
// for compatiblity with Arduino Due and Teensy 3.0 and maybe others?
#define PROGMEM
#define PGM_P  const char *
#define pgm_read_byte(addr) (*(const unsigned char *)(addr))
#define pgm_read_word(addr) (*(const unsigned char **)(addr))
#define strcpy_P(dest, src) strcpy((dest), (src))
#endif
#include <string.h> // for strcpy_P or strcpy
#include "Time.h"
 
// the short strings for each day or month must be exactly dt_SHORT_STR_LEN
#define dt_SHORT_STR_LEN  3 // the length of short strings

static char buffer[dt_MAX_STRING_LEN+1];  // must be big enough for longest string and the terminating null

const char monthStr1[] PROGMEM = "January";
const char monthStr2[] PROGMEM = "February";
const char monthStr3[] PROGMEM = "March";
const char monthStr4[] PROGMEM = "April";
const char monthStr5[] PROGMEM = "May";
const char monthStr6[] PROGMEM = "June";
const char monthStr7[] PROGMEM = "July";
const char monthStr8[] PROGMEM = "August";
const char monthStr9[] PROGMEM = "September";
const char monthStr10[] PROGMEM = "October";
const char monthStr11[] PROGMEM = "November";
const char monthStr12[] PROGMEM = "December";

PGM_P monthNames_P[] PROGMEM = 
{
    "",monthStr1,monthStr2,monthStr3,monthStr4,monthStr5,monthStr6,
	monthStr7,monthStr8,monthStr9,monthStr10,monthStr11,monthStr12
};

matthijskooijman, 2015-10-13:
This is not a compiler issue, you have to actually declare the array as const. Note that you now have `const char * monthNames_P[] PROGMEM`, which means you have an array, in PROGMEM containing pointers-to-const-char. But the pointers themselves aren't constant. Try:

     PGM_P const monthNames_P[] PROGMEM = 

I'm closing this issue, as this is a problem that should be fixed in the code, not the compiler.

closed 2015-10-13 by matthijskooijman

cmaglie, 2015-10-14:
There is also a small FAQ here: https://github.com/arduino/Arduino/wiki/1.6-Frequently-Asked-Questions#errors-related-to-avr-progmem-changes