arduino/Arduino #3977: Ethernet2 client source port causes unreliable connect through some networks

guentheratge, 2015-10-13:
The change of the source port (EthernetClient::_srcport) to 1024 instead of the previous 49152 in the original Ethernet library causes some networks to reject the packets as suspicious.  It is particularly blocked by some of the Amazon AWS firewalled environments and Cisco IDS and IPS.  Switching the source port back to the IANA recommended ephemeral ports beginning at 49152 fixes this.  I'm unsure why this was changed to 1024 in Ethernet2, but this causes problems connecting to some hosts.

guentheratge, 2015-10-13:
This is how I patched my library and this allows my client to work with all servers:

diff -rU5 Ethernet2/src/EthernetClient.cpp Ethernet2-ag/src/EthernetClient.cpp
--- Ethernet2/src/EthernetClient.cpp	2015-09-18 12:50:14.000000000 -0700
+++ Ethernet2-ag/src/EthernetClient.cpp	2015-10-13 12:02:25.000000000 -0700
@@ -10,11 +10,11 @@
 #include "Ethernet2.h"
 #include "EthernetClient.h"
 #include "EthernetServer.h"
 #include "Dns.h"
 
-uint16_t EthernetClient::_srcport = 1024;
+uint16_t EthernetClient::_srcport = 49152;
 
 EthernetClient::EthernetClient() : _sock(MAX_SOCK_NUM) {
 }
 
 EthernetClient::EthernetClient(uint8_t sock) : _sock(sock) {

q2dg, 2015-10-13:
This product is not official. You won't be able to get any help from here, sorry. But thanks for your explained solution.

guentheratge, 2015-10-13:
Thank you.  My intention was to point out an issue with the library that comes with the Arduino IDE in the event that someone has a similar issue.  Do you know who maintains the Ethernet2 library, that I might get this fix put in it for everyone?

q2dg, 2015-10-13:
Well..."the others"

matthijskooijman, 2015-10-14:
@guentheratge, this library is not supplied with the Arduino IDE from arduino.cc, but only with the version from arduino.org (it's a bit hairy, but yes, there are two versions right now). I recommend you look at that site and report your issue there. I'm closing this issue here.

closed 2015-10-14 by matthijskooijman