arduino/Arduino #3981: Differences in binary/optimizations between Linux and Windows, Windows binary not working

jeskata, 2015-10-14:
I have a problem here - a project that works when it is compiled under Linux does not work when compiled under Windows.

Using Uno but preparing to migrate to a Mega1280.

I have my own modified/developed library for CC1101 radio module based on PanStamp and a sketch to use it to receive data from a measurement node.
Problem is simply that the binary compiled under Windows will not receive any packets. It gives no interrupts.

I have of course tried flashing on Linux and connecting the board on the Windows machine if there would be any RF disturbances, but no - it works perfectly.
Also the binary size differs - almost 2 kB larger on Windows but it varies depending on version. I have removed all the directories created under AppData/Temp.

On Linux (LUbuntu) I have 1.0.5+dsfg2-2.
On Windows(7) 1.0.5-r2, also tried 1.6.2 and 1.6.5-r5 (newest). 1.0.5-r2 and 1.6.5-r5 only have this particular additional library.

Is there some optimization differing here, what is causing this?

It is based on this http://labalec.fr/erwan/?p=497 , and on both versions I can read out the PARTNUM/VERSION registers in the setup function. But it never gives any interrupt when a packet should arrive. The interrupt pin is active, can be tickled when floating.

I have made the two builds and taken an objectdump (objdump.lst in respective directory). Builds together with sketch and library at
http://users.abo.fi/jskata/Arduinoproblem/

Hopefully the objdumps say more to someone else here than it does to me...

NicoHood, 2015-10-14:
It would be also interesting to see what the current nightly gives, since they changes the whole compiler structure there. You might want to try this as well.

Sounds very weird...
Can you try to reduce the problem to a smaller testcase?

matthijskooijman, 2015-10-14:
@jeskata, you should probably stop using 1.0.x versions for this. That version is legacy, and we can't really spend the time required to debug this for that version. Also, the gcc versions shipped with 1.0.x were somewhat random and they were taken from different places. With 1.6.x, all gcc versions are built by Arduino, from identical sources, so they should be more similar.

I looked at the files you supplied, and the .elf files are *really* different, so I suspect you compiled these with 1.0.x? Could you do the same with 1.6.5, or, even better, the hourly build? Hopefully that either solves the problem, or gives a more meaningful diff.

jeskata, 2015-10-14:
OK, the application was originally developed using 1.0.5 on Linux so naturally I tried to verify using the same version. Also, Ubuntu and Gentoo still only have 1.0.5 in repository...

I am somewhat suspecting the ISR cc1101signalsInterrupt is too long, could that be? Well it is too long for an ISR but are there technical limitations how long it may be or how long it may take?
I will attempt to reduce the sketch to a minimum and make the ISR a flag only, as a next step when I have time. Driver kept as it is.

matthijskooijman, 2015-10-15:
@jeskata, installing a newer version of the IDE is as simple as downloading and unpacking the tarball from arduino.cc, no installation required at all (so multiple versions can peacefully co-exist as well).

> I am somewhat suspecting the ISR cc1101signalsInterrupt is too long, could that be? Well it is too long for an ISR but are there technical limitations how long it may be or how long it may take?
I will attempt to reduce the sketch to a minimum and make the ISR a flag only, as a next step when I have time. Driver kept as it is.

An ISR can be as long as you'd like AFAIK, there are no size differences. However, while the ISR runs, interrupts are disabled, so if it runs for a long time, there will be increased latency for other interrupts (and they might even get lost if the same interrupt occurs *twice* while interrupts are disabled, since only one interrupt of each type can be queued).

jeskata, 2015-10-16:
This seems to be solved! There is a serious flaw in the example at http://labalec.fr/erwan/?p=497 ,

byte syncWord = 199;

And https://code.google.com/p/panstamp/source/browse/trunk/arduino/libraries/panstamp/cc1101.cpp
/**
 * setSyncWord
 *
 * Set synchronization word
 *
 * 'syncH'      Synchronization word - High byte
 * 'syncL'      Synchronization word - Low byte
 * 'save' If TRUE, save parameter in EEPROM
 */
void CC1101::setSyncWord(uint8_t syncH, uint8_t syncL, bool save)
{
  if ((syncWord[0] != syncH) || (syncWord[1] != syncL))
  {
    writeReg(CC1101_SYNC1, syncH);
    writeReg(CC1101_SYNC0, syncL);
    syncWord[0] = syncH;
    syncWord[1] = syncL;
    // Save in EEPROM
    if (save)
    {
      EEPROM.write(EEPROM_SYNC_WORD, syncH);
      EEPROM.write(EEPROM_SYNC_WORD + 1, syncL);
    }
  }
}

/**
 * setSyncWord (overriding method)
 *
 * Set synchronization word
 *
 * 'syncH'      Synchronization word - pointer to 2-byte array
 * 'save' If TRUE, save parameter in EEPROM
 */
void CC1101::setSyncWord(byte *sync, bool save)
{
  CC1101::setSyncWord(sync[0], sync[1], save);
}

This led to the second byte depending on compile and optimizations turning random, it is supposed to be zero. Sometimes it has then been zero and those builds have worked.

Changed to
byte syncWord[2] = {199, 0};

Now I can move to a recent Arduino on all platforms!

facchinm, 2015-10-20:
Perfect :wink: Marking as solved

closed 2015-10-20 by facchinm