arduino/Arduino #3987: IDE 1.6.5 Windows 8.1

Aliwalio, 2015-10-16:
Having upgraded from 1.0.5 to 1.6.5 I have had to revert to old IDE  as new version seems unusable. Have tried 1.6.0 with same results.

With both these new versions the WMI Provider Host process hogs 30% of CPU and the IDE stalls. Seems also to have something to do with plugging in the USB cable to board. Also, intermittently, port number is not recognised or lost.

Some unkind comments re this problem also here : https://www.reddit.com/r/arduino/comments/2vbcm7/arduino_ide_16_is_released_download_it_now/

Having reverted to 1.0.6 everything works fine.

ffissore, 2015-10-26:
It looks like a windows problem. Try checking this http://answers.microsoft.com/en-us/windows/forum/windows_8-performance/wmi-provider-host-constantly-disrupting/8e1ead34-d25f-4ee5-8eb1-7a6afe3c0696?auth=1
Disabling your antivirus is also an option, although not really feasible for windows users

closed 2015-10-26 by ffissore

Aliwalio, 2015-10-27:
Mille grazie Federico. Very kind of you to respond so quickly.

I will try the workarounds suggested when I get a moment and who knows the Windows 10 upgrade (eventually) might fix anyway…..

I see also that you have earmarked this issue for 1.6.6 so good luck with it.

Regards

Alistair

Spence Clarke S.L., Tel: +34 952 822 943 Fax: +34 952 827 779

SPENCE CLARKE & CO
30 YEARS IN SPAIN - 1985 - 2015

Spence Clarke & Co supports the charity Motor Neurone Disease Association (UK). They are currently helping to support and promote a longstanding fundraiser for this charitable organisation, namely www.justgiving.com/mndassoc/donate<https://www.justgiving.com/mndassoc/donate>

PRIVACY & CONFIDENTIALITY NOTICE
Information contained in this email and any attachment is confidential and intended for the addressee only. If you are not the intended recipient please notify us immediately. Any dissemination, distribution, copying or use of this information without our prior consent is strictly prohibited. The views of the author may not necessarily reflect those of Spence Clarke S.L. If you have any doubts as to the authenticity of this e-mail please contact either the sender or the company directly.

VIRUS WARNING
The contents of any attachment to this email may contain software viruses that could damage your own computer system. While Spence Clarke S.L. has taken reasonable precautions to minimise this risk, it cannot accept liability for any damage that you suffer as a result of software viruses. You should carry out your own virus checks before opening any attachment.