arduino/Arduino #3992: Ethernet example code doesn't work

richw42, 2015-10-18:
At the URL
file:///C:/Program%20Files%20(x86)/Arduino/reference/arduino.cc/en/Reference/ServerAvailable.html
I see the following code:
if (client == true) {
    // read bytes from the incoming client and write them back
    // to any clients connected to the server:
    server.write(client.read());
  }
In my tests, the (client == true) test is never true.
Changing the test to (client) fixed the problem for me.
This same code is found in other places in the documentation.

richw42, 2015-10-18:
Hmmm, maybe it makes a difference:
Downloaded install file is arduino-1.6.5-r5-windows.exe
Using Arduino Ethernet board.
Windows 7-64

agdl, 2015-10-27:
@richw42 unlucky the documentation provided with the downloaded IDE has some update problem. Please always refer to the online documentation that is always updated and maintained.

agdl, 2015-11-17:
I'm closing this

closed 2015-11-17 by agdl