arduino/Arduino #3993: "Link status" missing

ibantxo, 2015-10-18:
In the shield, there is some status leds: PWR, LINK, 100M...

How could I get the "LINK" status from my code?

ibantxo, 2015-10-19:
I have found the ISSUE number 855: Ethernet library should doesn't notice if the physical layer is disconnected (https://code.google.com/p/arduino/issues/detail?id=855&can=1&q=w5100%20link)

That is my problem!!

As they say, reading the W5100 datasheet v1.2.2 there is no register corresponding to the link status. But there is a led for it!!!

ibantxo, 2015-10-19:
Maybe reading one W5100 internal register via SPI from arduino...

NicoHood, 2015-10-20:
Maybe you should get the ethernet shield schematic and look where the led is conencted to. It must be connected to the W5100. And if your datasheets says there is no register to read the data, only the pin lets you know the state. So you can solder a wire to the led or just... give up. (I never read the datasheet, just answered based on the given information)

cmaglie, 2015-10-20:
Duplicate of #855.

Unfortunately it's not possible, the W5100 has no registers that reports the status.

closed 2015-10-20 by cmaglie