arduino/Arduino #3994: Reference

hgBranzell, 2015-10-18:
When you install the Arduino sofware on your computer you also get a folder named Reference that contains all the html pages that constitute the Reference. One of them is the main page, HomePage.html with links to everything else. When you try open this page it fails. It opens but never gets ready and the links are not clickable. 

I am running Windows Vista with Google Chrome.

ffissore, 2015-10-26:
working on it

hgBranzell, 2015-10-26:
Good, not that I desperately need it to work, I am on line and that works.
But not everybody is, at least not all the time.

I copied the package to my system drive which is SSD, i e vary fast, but it
still took forever.

Let us see if you can reproduce it. Maybe it is only on my computer.

Cheers.

ffissore, 2015-10-26:
54ec785fd211c5c1142dc35995e832edd11095e3 should contain a fix. It will become available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-10-26 by ffissore

hgBranzell, 2015-10-26:
Good, I will test it for us! :-)