arduino/Arduino #3995: Arduino IDE uses wrong path for avr-g++

Avamander, 2015-10-18:
IDE version: 1.6.6 2015/10/14 10:26
OS: Ubuntu 15.04

I encountered this error:

     /bin/avr-g++: no such file or directory

I solved it with this command as the actual location is /usr/bin/ not /bin/ .

    for ff in `ls /usr/bin/avr-*` ; do ln -s $ff $(echo $ff | awk -F "/" '{ print $4 }') ; done

Now it says this:

    {runtime.tools.avr-gcc.path}/bin/avr-g++: no such file or directory

Uh, oh. How would I fix this?

facchinm, 2015-10-20:
Hi @Avamander , it looks like `{runtime.tools.avr-gcc.path}` is not expanded correctly in your environment, so the build system drops it and all the resulting paths are wrong.

Are you using a precompiled nightly or a custom one? In the first case, can you take a look inside `arduino-nightly/hardware/tools/avr/bin` and check if all the files exist with correct permissions?

Avamander, 2015-10-20:
I used precompiled nightly. The permissions were indeed wrong, I did not with to encounter more issues so I did a pure install. It now compiles.

closed 2015-10-20 by Avamander