arduino/Arduino #3996: Library Manager, update CRC

plinioseniore, 2015-10-18:
Hi,

I've messed-up a bit and used --force for a tag, so now it has a different CRC and the Library Manager doesn't complete the installation of the library.

Could you please reset the CRC for 

```
    {
      "name": "souliss",
      "version": "7.1.0",
      "author": "Souliss Team",
      "maintainer": "Souliss Team",
      "sentence": "SmartHome Networking Framework",
      "paragraph": "Build a network of multiple nodes over Ethernet, Wireless and RS485 for your smart home",
      "website": "http://souliss.github.io",
      "architectures": [
        "avr",
        "esp8266"
      ],
      "types": [
        "Contributed"
      ],
      "url": "https://github.com/souliss/souliss/archive/v7.1-friariello.0.zip",
      "archiveFileName": "souliss-7.1.0-github.zip",
      "size": 621126,
      "checksum": "SHA-256:b7a44bcffb1ae1fb6cc855e5a43d6e446dffcfc1d9be47b89172432fba5e3bf2"
    },
```

Thanks

matthijskooijman, 2015-10-18:
I suspect this happens automatically during the weekly cleanup run on sunday evening, so could you please check again tomorrow and report if the problem still exists then?

plinioseniore, 2015-10-18:
@matthijskooijman just had a try, the JSON has been updated but the checksum is still the old one. At which time is supposed to run this clean-up?

matthijskooijman, 2015-10-19:
Not sure, on sunday evening (CEST, probably) I believe. Should have ran now, I think.

plinioseniore, 2015-10-19:
Unfortunately I still get a CRC error, I've checked that the JSON file has been updated by the IDE.

cmaglie, 2015-10-20:
It seems that you pushed the release right after the weekly cron job. I've triggered it manually and now the CRC seems fixed.

closed 2015-10-20 by cmaglie