arduino/Arduino #3999: SD.exists and SD.remove bugs

MyNick, 2015-10-19:
For some unkown reason those function receive the file path as char* and not const char*. Moreover, if you pass it const char* then a seg fault happen (the program just stop and reset over and over). Some how the compiler let you do it but its another bug.

cmaglie, 2015-10-20:
I see that the prototypes are:

```C++
  boolean exists(char *filepath);
  boolean exists(const String &filepath) { return exists(filepath.c_str()); }
```

changing the prototype to accept a `const char *` instead of a `char *` is a good thing to do, BTW I don't see how the compiler can accept a `const char *` here, maybe it is silently converted to a `String`? Can you provide a small test case? Which board are you using?

MyNick, 2015-10-21:
On my arduino uno:
http://pastebin.com/EH3vEkae - Working
http://pastebin.com/Fy0pa76p - Not working, never reach the after. Plus i have a test which open a file and write the start message into the file. After one sec i have a few hundred on "start" lines.
The only difference is that i added const

matthijskooijman, 2015-10-21:
Weird indeed, I would expect a compiler error. Silently converting to `String` could be what happens, though that doesn't explain why it breaks either....

Ivan-Perez, 2015-10-21:
By chance, one hour before you opened this issue I made a pull request fixing this: https://github.com/arduino/Arduino/pull/3998. You can use the builds on that PR to check if that fixes your problem.

By the way, I didn't have any problems before, the compiler only showed me a warning about converting a `const char*` into a `char*`. All sketches worked for me before as they're working now after adding those consts (I'm using an Arduino Mega board).

matthijskooijman, 2015-10-21:
@MyNick, could you see if the builds generated in #3998 fix your problem?

MyNick, 2015-10-21:
How do i do it? how can i choose to what link to in the arduino software or how can i compile it from the cmd?

matthijskooijman, 2015-10-21:
@MyNick Arduinobot has kindly built a version of the Arduino IDE with the patches from #3998 applied already, see this comment: https://github.com/arduino/Arduino/pull/3998#issuecomment-149735955

In that comment, select the link that matches your platform, download the corresponding file and unpack it (can be anywhere, no install required). After unpacking, you will find an Arduino.exe file inside (for Windows), which you can doubleclick to start.

MyNick, 2015-10-21:
The pull request solved it. Thank you. Ill wait for the official release

cmaglie, 2015-10-22:
fixed with #3998

closed 2015-10-22 by cmaglie

sauerburger, 2015-11-30:
I had the same problem that the arduino crashes when calling `exists` with `const char*`. I would really would like to know why the compiler didn't complain. Because if I write a function with a `char*` argument and pass `const char*`, I get an error.

treii28, 2016-08-22:
I'm getting a crash on SD.exists with a string when pointing it at a directory in the root dir. I've tried upper and lower case, leading '/' no leading slash, trailing '/' no trailing slash. Same results with all variations. The directory does exist.

treii28, 2016-08-22:
I was getting the same problem trying to use a string as a filename. I tried a variety of things from using .c_str(), to strcpy with a char* buffer. The only thing I could eventually get to work is converting the string to a char using toCharArray on the string.