arduino/Arduino #4002: Conflicting type for first argument of utoa with stdlib.h declaration

jlbirccyn, 2015-10-20:
Hello

In `hardware/arduino/sam/cores/arduino/itoa.h` utoa is declared as follow:

```
extern char* utoa( unsigned long value, char *string, int radix ) ;
```

in a gcc 4.9.3 for arm distribution, it conflict with the declaration of the built-in utoa:

```
char *	_EXFUN(__utoa,(unsigned, char *, int));
```

(_EXFUN is defined as follow `#define	_EXFUN(name, proto)		__cdecl name proto
`)

cmaglie, 2015-10-20:
Is this specific for arm-gcc 4.9.3 or for other versions as well?
Can you provide a test sketch?

jlbirccyn, 2015-10-21:
Seems to be with 4.9.3. I do not have the error with 4.9.2. No need to provide a sketch. Go in `hardware/arduino/sam/cores/arduino` and type `arm-none-eabi-gcc -c -o test.o -I. WString.cpp` (or whatever name your compiler have, provided it is 4.9.3). You can get a 4.9.3 from https://launchpad.net/gcc-arm-embedded

jsanchezgarcia, 2016-03-03:
any updates on this?

neojou, 2016-03-22:
I have the same problem when trying to upgrade toolchain from 4.8.3 to 5.2.1.

patrickelectric, 2016-03-31:
Same error here :disappointed:

jlbirccyn, 2016-04-06:
Just sent a pull request to fix the problem

cmaglie, 2016-10-17:
Should be fixed with #5480.

Please reopen, or furhter comment, if not.

closed 2016-10-17 by cmaglie

clovisf, 2016-11-09:
I have tested (**and confirm it works**) this fix with Arduino IDE Nightly Build 1.6.13 on windows 10 using a third-party board with ATSAMD21G18A, after burning the latest Arduino Zero bootloader to it using ATMEL Studio 7.0