arduino/Arduino #4003: http://playground.arduino.cc/Code/CascadingRegisters is an empty page

alexandr-, 2015-10-20:
valid link --- eg from http://playground.arduino.cc/Code/

cmaglie, 2015-10-20:
I don't know if it's possible to delete a page, maybe just removing the link from "Code" would be sufficient in this case?

alexandr-, 2015-10-20:
i dont know why it is gone. i was actually looking for that information via http://playground.arduino.cc/Code/ShiftRegister595  (see faq)

per1234, 2015-10-20:
You can delete a page from the playground by editing and replacing all the text with the word `delete`. I have done so after first checking the history to make sure there was no previous content that would be lost(there was none). The http://playground.arduino.cc/Code/ page is automatically generated so the CascadingRegisters page has also been removed from that page. I removed the link from the ShiftRegister595 page.

cmaglie, 2015-10-20:
> You can delete a page from the playground by editing and replacing all the text with the word `delete`

Ah! cool, nice to know, thanks @per1234 !

closed 2015-10-20 by cmaglie