arduino/Arduino #4004: New library version is not published

bportaluri, 2015-10-20:
Hi, I'm the author of ALA library (https://github.com/bportaluri/ALA).
Version 2.0 is already available in the Library Manager.
However, I have published version 2.2.0 on GitHub several days ago but I still do not see the new version in the Library Manager (https://github.com/bportaluri/ALA/releases/tag/v2.2.0).
I can see the updated release has been pulled by the hourly task: http://downloads.arduino.cc/libraries/library_index.json
I have also updated the library.properties file.

Anything wrong?

Regards

matthijskooijman, 2015-10-20:
When I open my IDE, I can install version 2.0.0, 2.1.0 and 2.2.0. I wonder if the packages JSON is downloaded correctly by your IDE? You can check in ~/.arduino15/ for the library_index.json downloaded by the IDE (on Linux, not sure about the Windows or OSX paths).

Note that you have a 2.2 and 2.2.0 tag, but both point to the same commit and it seems the library manager picked up the 2.2 tag (ignoring the 2.2.0 tag as duplicate), which should be ok.

bportaluri, 2015-10-20:
Ok I have forced a rebuild of the IDE settings deleting C:\Users\Twim\AppData\Roaming\Arduino15 folder.
Now I see the three versions and everything works fine.
Thank you.

BTW I have deleted v2.1 so I'm expecting it will be removed from the list this saturday right?

matthijskooijman, 2015-10-20:
Great! I'll close this issue, since it seems it was only temporary.

Deletion of versions happens once per week, IIRC on sunday.

closed 2015-10-20 by matthijskooijman