arduino/Arduino #4007: Due TWI1 pins  (2nd i2c port): are there integrated pullups on-board or not?

VogonJeltz, 2015-10-21:
hello,
about Due TWI1 pins (2nd i2c port): are there integrated pullups on-board or not?

VogonJeltz, 2015-10-22:
why is there no one who can comment and fix this effing issue?

mikaelpatel, 2015-10-22:
Seems like you need some help to search info. I found this (in 10 us :-) here: https://www.arduino.cc/en/Main/ArduinoBoardDue

> TWI 1: 20 (SDA) and 21 (SCL)
> TWI 2: SDA1 and SCL1.
> Support TWI communication using the Wire library. SDA1 and SCL1 can be controlled using the 
> Wire1 class provided by the Wire library. While SDA and SCL have internal pullup resistors, SDA1 
> and SCL1 have not. Adding two pullup resistor on SDA1 and SCL1 lines is required for using Wire1.

Cheers!

VogonJeltz, 2015-10-22:
thank you, but then I wonder why the Mega with disables pullup resistors work as a i2c slave (to a Lego mindstorms EV3) and the DUE don't - neither by Wire not by Wire1.

http://www.mindstormsforum.de/viewtopic.php?f=25&t=8673#p67709

In this case it must be a hardware bug which needs to be fixes.

mikaelpatel, 2015-10-22:
Can you get the Due to work as an I2C slave at all? For instance with a Mega as master. It does not necessary have to be a hardware issue. Have you tried debugging the Due I2C code? Checking the state machine etc. 

Thanks for the poetry :)

mikaelpatel, 2015-10-22:
BW: Is there a difference in voltage? The pullup resistors on the Lego Mindstorm might be to large and/or expects 5V? What is the spec? The Due is a 3V3 board. Is it not?

VogonJeltz, 2015-10-23:
the EV3 is at 3.3V level but has a 5V protection. All new EV3 sensors also work at 3.3V level.

2 DUEs work by i2c, 1 as a master, 1 as a slave (Wire-Wire, Wire-Wire1, Wire1-Wire).
(did not try Wire1-Wire1).

The dexter industries example works with an Uno, basically it's similar to my approach using a Mega.

mikaelpatel, 2015-10-23:
@VogonJeltz 

Good work! You will need a scope or at least a logic analyzer for the next step. So far this looks like a hardware problem in your setup. There is no evidence that there is a problem with the Due board or the software then. 

Have you tried a level shifter with the correct pullups on both side? 

Cheers!

VogonJeltz, 2015-10-23:
no, I dont have level shifters and I also don't have a scope, I am not experienced in electrical stuff. Just plugging wires from one header to the other is ok of course.

mikaelpatel, 2015-10-23:
@VogonJeltz 
I think that this issue is answered now and should be closed. Good luck with your project!

VogonJeltz, 2015-10-23:
I'm not quite sure if everything is answered.
First of all I would like to try to disable the internal Due pullups on Wire by software, like I did for the Mega (patch in ...\avr\lib.c ) .
How can I do this for the Due?

NicoHood, 2015-10-23:
This is not a support forum. It really looks like a hardware issue, so arduino IDE developers cannot really help you. You might want to ask in the forums or chat about this specific project question.

A Level shifter (logic level converter) is 1$ and a logic analyzer (salae china clone) is 8$. Both can be really useful at any time/project.

VogonJeltz, 2015-10-23:
the design of twi.c in the avr subfolder is there by default, but it can only disable AVR pullups, no ARM pullups.
So there has to be changed something in the ARM libs, and this is a question or a request to the developers, not to end users in either user forum.

NicoHood, 2015-10-23:
As said above from @mikaelpatel we do not have your hardware and cannot help you debugging your hardware. Grab a logic level converter and a logic analyzer and provide us more information.

And if you need to change the "arm" lib, go ahead, its open source. The due is SAM by the way, no ARM.

The forums are still a better place, since mostly noone checks the github issues, except some interrested developers. You have a better change to ask there or in the IRC chat. From time to time there are very experienced users.

Another idea would be to ask @sandeepmistry who created a lot of PRs with TWI fixes. He is the guy to ask I guess.

Good luck.

VogonJeltz, 2015-10-23:
which part was ununderstandable?
my question was not about hardware debugging, 
it simply about disabling Due-pullups on Wire by a software command.
Just like for AVRs

This is my issue.

mikaelpatel, 2015-10-23:
@VogonJeltz 
I took a few minutes to browse through the SAM TWI code. I was a bit curious to how it was done as I mainly work on my own framework (https://github.com/mikaelpatel/Cosa)

Actually I could not find anything that looked like enabling of pullup - that assumption seems to be wrong for the SAM. In any case you could write a few lines of code and check the pin mode after initiating as TWI slave. I don't have an Arduino Due so I am sorry that I cannot help with that. I would also like to suggest that you check the ATMEL SAM documentation for the TWI. Guess there is some interesting info there. 

Keep up the poetry!

NicoHood, 2015-10-23:
SDA/SCL0-3 are connected to a pullup by hardware. (pin 7/8).
The default pins (above D13/AREF)for I2C dont seem to be connected to any HW pullup.

Just use a multimeter or @mikaelpatel s suggestion.

VogonJeltz, 2015-10-23:
yes, also on the Mega SDA/SCL are connected to hardware pullup resistors, but they can be enabled or diasbled by software command in twi.c.

The feature request to the developers was:
If not implemented FTM: please implement it for the future release.
Architecture: SAM
Library: Wire
Type: improvement

NicoHood, 2015-10-23:
Well it has in chip pullups and on PCB. Maybe the volt level is a problem or the Pullup is too strong/too weak.

The devs have recognized your issue and will respond if they feel to respond.

mikaelpatel, 2015-10-23:
Now I think there is some confusion about hardware pullup resistors (internal or external). On the Mega (https://www.arduino.cc/en/uploads/Main/arduino-mega2560_R3-sch.pdf) there are no external hardware pullup resistors. The AVR device driver code uses the internal port pullup resistors to protect the pins in TWI mode. These are large >10 K but apparently affect the Lego Mindstorm. 

The Due (https://www.arduino.cc/en/uploads/Main/arduino-Due-schematic.pdf) on the other hand has external pullup resistors on Wire0 1K5 to VCC (3.3V). There are no external pullup resistors on Wire1. 

@VogonJeltz I think you have got this mixed up. Please check the schematics and the device driver code.

NicoHood, 2015-10-23:
Of course there are! RNB1 and RNC1, 10k pullup to 5V top middle next to the isp header

mikaelpatel, 2015-10-23:
@NicoHood 
Found them! But they are hardly controlled by software as @VogonJeltz claims :-). With the internal pullup in parallel this gives approx. 6-8K. Deactivating the internal pullup gives these 10K pullups to 5V which seems to work with the Lego Mindstorm. But this does not help much with the Due? It puts the record straight with Mega though. 

Cheers!

VogonJeltz, 2015-10-23:
so can you confirm that there are DUE-pullups both on Wire and Wire1 (either if internal or external would not matter)?

That would explain the issue, because pullups on the bus have to be at least 32k... up to 82k.
10k is too low for EV3, it pulls up the wires too strongly . if not disabled, no communication to EV3 will be possible.

remark:
on the Mega instead, I once was told that there are no internal baut external pullups on the board, somewhere in the corner close to pins 18-22 - but actually that does not matter because they can be disabled by software completely which makes the Mega work with additional "extra-external" 47k  resistors

anyway, again, this statement would be the crucial question:
can one confirm that there are DUE-pullups both on Wire and Wire1 (either if internal or external would not matter)?

VogonJeltz, 2015-10-24:
next, 
if there indeed are pullups anywhere on the Due board 
- how can you disable by software 
pullups on Wire
pullups on Wire1
?

peabo7, 2015-10-24:
> On October 24, 2015 at 1:40 AM VogonJeltz <notifications@github.com> wrote:
> 
> next, 
> if there indeed are pullups anywhere on the Due board 
> - how can you disable by software 
> pullups on Wire
> pullups on Wire1
> ?

What does this have to do with the Arduino IDE?

You need to consult the schematic and data sheets (noted previously IIRC).

Try something out.  You are not as simple as you claim.  If you find something
relevant to Arduino, find out how to submit a pull request.

VogonJeltz, 2015-10-24:
which part of this thread didn't you understand? 
in the rduino-IDE there is a file called twi.c for avr as I already repeatedly had posted, and with a twi.c patch it worked, search the post above!

Now the request is about a similar API functionalty for the Due, is that so hard to understand?

NicoHood, 2015-10-24:
But we cant help you debugging your hardware problem without any hardware. You need to give us some information, as requested here:
https://github.com/arduino/Arduino/issues/4007#issuecomment-150532579
https://github.com/arduino/Arduino/issues/4007#issuecomment-150590042

peabo7, 2015-10-24:
> Now the request is about a similar API functionalty for the Due, is that so hard to understand?

Not a problem, can you work on it?

We're all volunteers here: help yourself.

VogonJeltz, 2015-10-24:
peabo, 
Man, if I could I had not asked.
I am a simple Arduino beginner and hobby programmer, I have no idea of electonics or C compilers, I just use the Arduino Sketch because it's so simple and I dont have to care about all this techic mess.

VogonJeltz, 2015-10-24:
@Nico, what di you want to know?
please simple words, short sentences, I need to use Google translate for it.
Or post in German, that would be best.
I can only report the issue and the twi.c request for the Due.

NicoHood, 2015-10-24:
Ich bin nicht der King in Sachen I2C. Das problem ist, dass wir dir einfach kaum helfen koennen, ohne hardware. Wenn sich die devs nicht mit einer Loesung melden, musst du wohl oder uebel ran. Und wahrscheinlich kann dir da auch niemand helfen, WEIL es ein HW und nicht SW problem ist.

Was er hier vorgeschlagen hat https://github.com/arduino/Arduino/issues/4007#issuecomment-150532579 war, dass du dir einen logic analyzer nimmst und schaust, was auf dem port abgeht. Raten bringt nichts. Nachschauen. Wie gesagt kostet so ein Teil aus china auch kaum was.

Hier https://github.com/arduino/Arduino/issues/4007#issuecomment-150590042 schreibt er, dass er in dem code nichts finden konnte, was einen pullup aktiviert. Es muss nicht zwanglaufig beim due auch soetwas geben, das sind ganz andere processor architektueren. Wenn dus wirklich wissen willst, musst du eben ins datenblatt schauen oder eben hardwareseitig messen. Ebenso hat er vorgeschlagen mal den pinmode auszulesen, nachdem der I2C slave initialisiert wurde. Ich schaetze mal es ging ihm darum zu sehen ob der pin high oder low ist. Einfacher gehts eben mit messen. Eventuell ist der pullup so stark, dass einfach kein signal durchkommt.

Und da wie oben erwahnt es wahrscheinlich auf ein HW problem hinauslauft musst du einfach einen level converter mal testen (1€) oder mit nem logic analyzer messen (8€). Wir haben keinen mindstroms hier rumliegen, noch die entsprechenden connector, noch die zeit dazu.

Wie gesagt am besten ist immer noch im chat zu fragen, weil da die bastler rumduedeln und hier mehr die entwickler. Wobei die auch sehr beschaeftigt sind. Kann ja wirklich sein, dass es ein bug ist etc. Aber im open source geschaeft bist du dazu angehalten den fehler selbst zu finden oder bei der loesung zu helfen. Letztendlich ist es so, und du kannst kaum was machen. Wenn du linux als OS nutzt, wirst du das auch da merken. Mal mehr mal weniger. Andererseits bekommst du natuerlich auch viel zurueck, wo andere etwas einbauen, was du bereits alles schon nutzt und dir gar nicht auffaellt. Aber naja darum gehts nun nicht.

Und bitte poste jetzt nicht in deutsch, das versteht hier wahrscheinlich keiner. Und da ich so gut wie gar keine ahnung von I2C habe kannst du mich auch nicht mehr Fragen, als wie du versuchen koenntest das Problem selbstaendig zu loesen. Keiner will dir boese oder dich aergern. Nur ohne deine Hilfe koennen wir das nicht loesen.

VogonJeltz, 2015-10-24:
Sokommen wir nicht weiter, ich habe kein elektronisches Equpment außer Jumper-Kabel und 1 Breadboard.

ich brauche erstmal eine ganz einfache, klare Antwort auf die alleresrte Gretchen-Frage (offensichtlich sind 3 Fragen auf einmal zu viel):

Gibt es nun wo auch immer eingebaute oder dazugelötete Pullups auf dem 2. I2C-Port auf dem Due, natürlich)?

NicoHood, 2015-10-24:
Nach Schaltplan hat nur der eine port (pin 7 + 8) HW pullups. Alle haben aber (vermute ich mal) interne pullups, im chip direkt.

VogonJeltz, 2015-10-24:
ok.
dann muss man sie  software abschalten können, genau wie auf dem Mega.
Und zwar alle, sowohl auf dem 1. als auch auf dem 2. i2c-Port.

Also die 2. elemntare Frage:
wie kann man sämtliche auf dem Due verbauten Pullups softwaremäßig abschalten genau  wie auch bereits schon auf dem mega möglich?

Google trabslate:
So the 2nd elementary question:
how can you turn off all built on the Due pullups in software just like already already on the mega possible?

mikaelpatel, 2015-10-24:
@VogonJeltz 
I borrowed a Due from a friend and got it to work with EV3, Wire1 and level shifter with large pullup according to spec. You might have fried your Due Wire1 port when playing around?

VogonJeltz, 2015-10-24:
the Due Wire1 works when plugged to a 2nd Due, no issues, both as a master and as a slave,
but both Dues don't work on Wire1 with the EV3 as a slave.

mikaelpatel, 2015-10-24:
@VogonJeltz 
Then you are doing something else wrong. If you have limited knowledge of electronics it is easy to connect something wrong such as ground. Anyway there is no problem when using a level shifter and the correct pullup resistors on the EV3 side.

VogonJeltz, 2015-10-24:
But level shifter?
why level shifter?
EV3 is 3.3V, too!?.

mikaelpatel, 2015-10-24:
@VogonJeltz 
Sorry but I cannot help you more. I was curious about the EV3 I2C, did some reading and testing. That is all I have to offer. Have a nice day!

VogonJeltz, 2015-10-24:
anyway, maybe my 2 Dues are broken and it's a hardware damage then.
But I won't buy  2 new ones then, too expensive

NicoHood, 2015-10-24:
Maybe you tell him your setup and then he can solve the problem. I dont think both dues are broken.

VogonJeltz, 2015-10-24:
anyway, it would be nice if one could switch off all the Due pullups by software just  like it's already possible for the Mega.
(ceterum censeo)

VogonJeltz, 2015-10-24:
update:
I also tried with pullups as mikaelpatel wrote:

NO CHANCE, NO CONNECTION AGAIN!
Mega works, DUE NEVER !

peabo7, 2015-10-25:
> I am a simple Arduino beginner and hobby programmer

This is your opportunity to advance your knowledge.  Everyone here was once a beginner.  Beginners become experts by trying to solve their problems.  The AVR code is your starting point.  The schematic and the datasheet for the DUE tell you what your resources are.  Try making a version for Due.  When you run into problems, ask questions.

VogonJeltz, 2015-10-25:
come on, Please spare me your conceited and  lecturing attitude!
It's a feature request to be  added to DUE lib improvements!
period!

cmaglie, 2015-10-26:
> 1st question: about Due TWI1 pins (2nd i2c port): are there integrated pullups on-board or not?

Pins "SDA" and "SCL" (number 20 and 21) have pull-up resistor on board.
Pins "SDA1" and "SCL1" does **NOT** have pull-up resistors on board.

This is explained here: https://www.arduino.cc/en/Main/ArduinoBoardDue

> Support TWI communication using the Wire library. SDA1 and SCL1 can be controlled using the Wire1 class provided by the Wire library. While SDA and SCL have internal pullup resistors, SDA1 and SCL1 have not. Adding two pullup resistor on SDA1 and SCL1 lines is required for using Wire1.

They can not be disabled via firmware because there are two physical resistors on board (you should de-solder the components from the board to **physically remove** the pull-ups).

The CPU on the Due board has pin pull-ups that can be controlled via firmware, but they are too weak (100KOhm) to be used in I2C bus.

VogonJeltz, 2015-10-26:
cmaglie,
thank you very much for clarification. So after all, the way the on-board-hardware-pullups on the Mega appear to be  constructed in another way, because they actually **can be disabled by software** - but this would absolutely not be possible for the Due, is this right?

matthijskooijman, 2015-10-27:
The internal pullups on the Mega (and most other AVR-chips) are smaller, typically just over 20k, which is usually small enough for I2C comunication.

For the Due, the current resistors are soldered in place, so no, there is nothing you can do in software to disable them. A future revision _could_ have a transistor added to enable/disable the pullups, but it seems unlikely that that extra complexity will be added.

I'm not sure why you actually want to disable these pullups, though? They are required for I2c operation, so unless you're running I2c with another device that runs at < 3.3V, I can't see any reason not to have these pullups?

VogonJeltz, 2015-10-27:
they have to be disabled to be interfaced as a slave to a "very special master device" (see posts above).
On the Mega the  board-pullups can be deactivated be patching the twi.c file (see above)
without deactiving the  board-pullups Mega don't work as a slave (see above),
after deactivating  the  board-pullups by software it works  (see above).

Now my question was, 
why is it possible to deactivate the board-pullups on the Mega by software via twi.c
and is it possible or is it ***absolutely irrefutably  impossible*** to do this on the Due in a similar way?

PaulStoffregen, 2015-10-28:
@VogonJeltz You are wasting the very valuable and limited time of many software developers with your selfish requests.  Please stop using Arduino's Github issue tracker for your personal project tech support.

DO NOT ask for improvements here regarding issues you do not understand.  Ask on the forum first.  You should never make these sorts of requests, without knowledge and experience, until other experienced people have confirmed the true nature of the problem.  The forum is meant for exactly that type of conversation.

For example, you should conduct this query on the forum.  Solicit experts to answer your questions regarding what is and is not possible by the hardware.  DO NOT POST HERE until after you have received confirmation from experts on the forum for your technical questions.

You are selfishly wasting precious time from people who do contribute to Arduino.  Please stop being so selfish and unproductive.

VogonJeltz, 2015-10-28:
Paul,
I didn't expect anything else than that from you.

VogonJeltz, 2015-10-28:
my question was,
why is it possible to deactivate the board-pullups on the Mega by software via twi.c
and is it possible or is it absolutely irrefutably impossible to do this on the Due in a similar way?

matthijskooijman, 2015-10-28:
I answered your questions a few comments ago, so I'm closing this issue. I'm with Paul on this one - please stop expecting other people to do your homework for you. This is not the place to ask questions, only to report bugs or suggest specific improvements.

closed 2015-10-28 by matthijskooijman

VogonJeltz, 2015-10-28:
no, you wrote something which was not the answer to my question.

my question was,
why is it possible to deactivate the board-pullups on the Mega by software via twi.c
and is it possible or is it absolutely irrefutably impossible to do this on the Due in a similar way?

If it's possibly in principal, please add this to the Arduino IDE.

matthijskooijman, 2015-10-28:
If you did not understand my answers, or think I misunderstood your questions, it is up to you to explain why you think this is so. Simply repeating your questions isn't going to help, since AFAICS I answered those questions exactly (though I'll likely stop investing time in you entirely...).

VogonJeltz, 2015-10-28:
no, you didn't pleas read closely what has been written!
closing this issue now as you didn't understand anything is really outrageous !!

NicoHood, 2015-10-28:
I love those type of people <3

VogonJeltz, 2015-10-28:
https://github.com/arduino/Arduino/issues/4007#issuecomment-150781100

VogonJeltz, 2015-10-28:
now what?

mikaelpatel, 2015-10-28:
@VogonJeltz 
Why did you not simply leave the issue closed the first time and contact me on the forum if you needed details on how I got it to work? Instead you just kept on "barking-up-the-wrong-tree". 

I will give you a final reference to all you need to know to get it working; http://stefanshacks.blogspot.se/2015/03/how-to-connect-lego-mindstorms-nxt_84.html
This is what I followed. This page has links to the schematics of all the Lego boards. Here you can see why http://www.lego.com/de-de/mindstorms/downloads. 

Please understand that you cannot disable hardware resistors on the board in software. You will need to remove them from the board.

VogonJeltz, 2015-10-28:
mikael,
thanks, your post was really helpful indeed, although it didn't work with my EV3 nevertheless, even using level shifters.

I finally abandoned all my tries to interface an EV3 to a DUE, it really makes no sense.

But I need to de-activate all pullups on all DUE i2c ports nevertheless, for different issues, even on SDA and SCL (the 1st i2c port).

That's why I've asked about completely deactiviating  pullups even on the first I2c.
SDA1 and SCL1 is out of interest in this respect as they already don't have any from the start.

After they appear to be hardware pullups on board, just like for the Mega, I would assume that it must be possible to deactivate in a similar software manner on the Due, just like it's possible for the Mega.

I really don't know how to make it more clear in English, regrettably.

[9 more replies]