arduino/Arduino #4008: Improvements on Arduino Zero product page

matthijskooijman, 2015-10-21:
I was reading the [Arduino Zero product page](https://www.arduino.cc/en/Main/ArduinoBoardZero) just now, and noticed a few things that I think are incorrect or could be improved. Here's the full list:
 - It says "The Atmel Embedded Debugger (EDBG) implements a JTAG interface in order to program the on-board SAMD21", but if I read the schematic correctly, it connects using SWD, not JTAG (JTAG is only available on the JTAG connector)? Similarly, later it says "uses the EDBG as a USB-to-JTAG chip".
 - It says "If using more than 12V, the voltage regulator may overheat and damage the board. The recommended range is 7 to 12 volts.", which applies to the linear regulators used on most Arduinos. However, AFAIS the Zero uses a switching regulator for the VIN->5V step, which shouldn't care so much about the input voltage?
 - It says "Pins that can be used for PWM output are: 3, 4, 5, 6, 8, 9, 10, 11, 12, 13 using analogWrite() function. " and later "PWM: 3, 5, 6, 9, 10, 11, and 13.", but those pin number lists don't agree.
 - AREF is mentioned twice in the list of pins.
 - It says "Either USB port can be used for programming the board, although using the Programming port is recommended due to the way the erasing of the chip is handled:", but that sentence is weird (you'd typically write either ... or but the or is missing here). Also, it talks about a way of erasing the chip, without clarifying what this means exactly and what the consequences of using the native port are.
 - It says "Unlike other Arduino boards which use avrdude for uploading, the Zero relies on bossac.", but I think that's only for the native port, the programming port uses openocd.
 - There is an ICSP connector, but can that actually be used for ICSP as well? Or is this ICSP stuff AVR-specific and is the connector just used for SPI?
 - The various serial ports are not clearly documented. AFAICS there are two hardware UARTS: `Serial`, connecting to the EDBG chip and `Serial1` connecting to pin 0 and 1, and one virtual serial port `SerialUSB` available through the native USB port (I haven't verified these, though).

per1234, 2015-10-22:
>It says "Either USB port can be used for programming the board, although using the Programming port is recommended due to the way the erasing of the chip is handled:", but that sentence is weird (you'd typically write either ... or but the or is missing here)

That seems correct to me. In this sentence either is used as a pronoun(meaning "either one") rather than a conjunction(either/or). So it's is basically saying "either the native USB port or the programming port" in less words which seems reasonable as the diagram right above the sentence shows both ports.

matthijskooijman, 2015-10-27:
> That seems correct to me. In this sentence either is used as a pronoun(meaning "either one") rather than a conjunction(either/or). So it's is basically saying "either the native USB port or the [...]

Ah, you're right, seems my head silently added a few words when I read that sentence. Thanks :-)

agdl, 2015-10-28:
@matthijskooijman please take a look now and let me know if it looks good

matthijskooijman, 2015-10-29:
Changes look good to me, I've crossed out the parts of my initial post that have been solved (note that they just look ok to me, I have not verified all info, like PWM pin list and input voltage range). Some of them still need to be solved, probably by someone who is more familiar with the details of the Zero board than me.

agdl, 2015-10-29:
About input voltage range i left what is written because there caps rated 25V and other rated little bit less so to play safe is better to stay in that range, about PWM I checked and it is right like is written now!

agdl, 2015-10-29:
Also about the ICSP it is used only for SPI and I stated it, about the Serials how do you suggest to state it?
About JTAG and connector you are right too, I missed to change it. It should be ok now!

agdl, 2015-11-17:
@matthijskooijman can this be closed now?

matthijskooijman, 2015-11-17:
I still have this on my TODO list, but I'm a bit busy now. I'll get back ASAP.

matthijskooijman, 2015-11-27:
Looks good now. Regarding the serial ports, how about adding a new section:

Serial ports
--------------
 - `Serial` is a hardware serial port internally connected to the EDBG chip and corresponds to the virtual serial port on your computer when you connect the Arduino Zero through the programming USB connector.
 - `SerialUSB` is a virtual USB serial port, which corresponds to the virtual serial port on your computer when you connect the Arduino Zero through the native USB connector.
 - `Serial1` is the hardware serial port connected to pin 0 and 1, which is free to use to connect to external serial devices.

agdl, 2015-12-03:
@matthijskooijman added thank you!

agdl, 2015-12-11:
@matthijskooijman I'm closing this, please reopen it if further improvements are needed!

closed 2015-12-11 by agdl