arduino/Arduino #4026: Export compiled Binary fails when board references variant from different vendor

per1234, 2015-10-24:
Using Arduino IDE 1.6.6 2015/10/23 11:42
- Select board that references a variant from a different vendor.
- Sketch > Export compiled Binary
- Error:
```
java.nio.file.InvalidPathException: Illegal char <:> at index 110: C:\Users\per\AppData\Local\Temp\untitled1839340130.tmp\sketch_oct24a\sketch_oct24a.ino.with_bootloader.arduino:mega.hex
	at sun.nio.fs.WindowsPathParser.normalize(WindowsPathParser.java:182)
	at sun.nio.fs.WindowsPathParser.parse(WindowsPathParser.java:153)
	at sun.nio.fs.WindowsPathParser.parse(WindowsPathParser.java:77)
	at sun.nio.fs.WindowsPath.parse(WindowsPath.java:94)
	at sun.nio.fs.WindowsFileSystem.getPath(WindowsFileSystem.java:255)
	at java.nio.file.Paths.get(Paths.java:84)
	at cc.arduino.Compiler.saveHex(Compiler.java:370)
	at cc.arduino.Compiler.saveHex(Compiler.java:346)
	at cc.arduino.Compiler.build(Compiler.java:148)
	at processing.app.Sketch.build(Sketch.java:1108)
	at processing.app.Sketch.build(Sketch.java:1083)
	at processing.app.Editor$BuildHandler.run(Editor.java:2011)
	at java.lang.Thread.run(Thread.java:745)
Illegal char <:> at index 110: C:\Users\per\AppData\Local\Temp\untitled1839340130.tmp\sketch_oct24a\sketch_oct24a.ino.with_bootloader.arduino:mega.hex
```
This is caused by the colon that is used in the reference syntax being included in the filename of the .hex file.

ffissore, 2015-10-26:
Which board have you selected? I see the bug but I would like to reproduce it myself, instead of blindly fixing it

per1234, 2015-10-26:
That particular error message came from a work in progress board but Adafruit Metro will do it:
https://github.com/adafruit/Adafruit_Arduino_Boards
Boards Manager URL: https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

ffissore, 2015-10-26:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

commit d8d70ee0 (2015-10-26): Windows: ":" is an illegal char for a file name. Fixes #4026
closed 2015-10-26 by ffissore