arduino/Arduino #4030: digitalWrite() is embarassingly slow on Due...

WestfW, 2015-10-27:
There's a relatively senseless use of Atmel Libraries that duplicate a lot of the effort.  The attached patch approximately doubles the speed of a pin-toggle loop...
(umm.  I can't attach a diff...)

--- wiring_digital.c~	2015-04-23 08:30:36.000000000 -0700
+++ wiring_digital.c	2015-10-26 20:59:25.000000000 -0700
@@ -84,7 +84,12 @@
   }
   else
   {
-    PIO_SetOutput( g_APinDescription[ulPin].pPort, g_APinDescription[ulPin].ulPin, ulVal, 0, PIO_PULLUP ) ;
+      if (ulVal) {
+	  digitalPinToPort(ulPin)->PIO_SODR = digitalPinToBitMask(ulPin);
+      } else {
+	  digitalPinToPort(ulPin)->PIO_CODR = digitalPinToBitMask(ulPin);
+      }
+//    PIO_SetOutput( g_APinDescription[ulPin].pPort, g_APinDescription[ulPin].ulPin, ulVal, 0, PIO_PULLUP ) ;
   }
 }

archenemies, 2016-06-06:
Seems like a useful patch

agdl, 2017-02-08:
This issue was moved to arduino/ArduinoCore-sam#16

closed 2017-02-08 by agdl

dsyleixa, 2018-07-21:
@westfw 
will this patch be fixed/integrated in the next official board core? (currently 1.6.11)?