arduino/Arduino #4032: Highlight "override"

NicoHood, 2015-10-27:
When you want to specify that you override a virtual method you use override as identifier. This should be highlighted.

closed 2015-10-27 by ffissore

peabo7, 2015-10-27:
I'm not conversant with C++11 yet, so I looked this up:

  http://en.cppreference.com/w/cpp/language/override

Depending on how this is implemented in the highlighter, there may be an issue:

**override** is an identifier with a special meaning when used after member function declarators: it's not a reserved keyword otherwise.

So if there is no check for that exact context (a tricky check, I imagine) it will be confusingly highlighted when used in a context where it has no special meaning.

NicoHood, 2015-10-28:
If you use a variable with the name "override" then its you fault. Its a limitation here for sure. There might be other words like that.

Thx for the fix

Chris--A, 2015-10-28:
@peabo7 
I think it should be highlighted. Then people may be discouraged from using it, apart from the circumstance where it does have special meaning.

This functionality would be similar to what already happens when using identifiers like   `begin`, `end`, `loop`, `setup`, which are all valid outside of their context which they are highlighted for.

peabo7, 2015-10-28:
@Chris--A @NicoHood OK, sounds like a good reason, thanks

commit ff248fc7 (2015-10-27): Added override and final as keywords. Fixes #4032