arduino/Arduino #4034: Problem with .git folder inside sketch

NicoHood, 2015-10-27:
I have a .git folder inside my sketch to track the changes.
Autosave needs to be disabled!

Try to compile with a change in the file and you will get an error. Save the sketch and compile again, it will work. I cannot post the error right now, the machine is offline currently. Tested on linux.

The IDE should probably ignore all .name folders/files for compiling. Also I get a warning for every library that contains a .git folder inside which is not very rare. This is a bit annoying too.

ffissore, 2015-10-27:
Please provide the error message. I've just tried compiling a Blink sketch with a .git folder and it works as expected

NicoHood, 2015-10-27:
You need to disable autosaving and add an unsaved change. Did you check that correctly?
I used multiple tabs with .h and .ino files.

Problem is, that I dont have internet right now, I will try to post the error as soon as possible.

NicoHood, 2015-10-28:
It was something like "cannot create temporary file /tmp/number/.git". Followed with a few java errors.
You should be able to recreate the issue very fast with the notes i added above.

ffissore, 2015-10-28:
Double checked autosaving and modified sketch. Still can't replicate. Waiting for full error message

NicoHood, 2015-10-30:
```
java.io.IOException: Unable to create folder: /tmp/arduino_122055ad210d4d42f9380af9ab708e2b/.git
	at processing.app.helpers.FileUtils.copy(FileUtils.java:63)
	at processing.app.Sketch.saveSketchInTempFolder(Sketch.java:1114)
	at processing.app.Sketch.build(Sketch.java:1105)
	at processing.app.Sketch.build(Sketch.java:1083)
	at processing.app.Editor$BuildHandler.run(Editor.java:2011)
	at java.lang.Thread.run(Thread.java:745)
Unable to create folder: /tmp/arduino_122055ad210d4d42f9380af9ab708e2b/.git
```

* Create a sketch
* Save
* git init
* compile
* change
* compile
* change
* compile
* error

You have to change the sketch twice to get the error for the first time.

ffissore, 2015-11-02:
Thanks @NicoHood Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-11-02 by ffissore

NicoHood, 2015-11-05:
Made a clean build from master today. Still got:
```
java.io.IOException: Unable to create folder: /tmp/arduino_b84c392175987be7c846c2e342599d63/.git
	at processing.app.helpers.FileUtils.copy(FileUtils.java:63)
	at processing.app.Sketch.saveSketchInTempFolder(Sketch.java:1114)
	at processing.app.Sketch.build(Sketch.java:1105)
	at processing.app.Sketch.build(Sketch.java:1083)
	at processing.app.Editor$BuildHandler.run(Editor.java:2011)
	at java.lang.Thread.run(Thread.java:745)
Unable to create folder: /tmp/arduino_b84c392175987be7c846c2e342599d63/.git
```

NicoHood, 2015-11-05:
Nevermind, I forgot to merge.

**However** I noticed that it compiles all files again after the first change. Maybe that is another issue. After that (which caused the bug to appear at that time) the already compiled files are used. So meaning after 2 compilations the optimization of the used files will take place, but not after the first compilation.

ffissore, 2015-11-06:
arduino-builder compares last modification time of source file and object file. Could this have changed somehow? maybe you `git checkout` to another branch or something?

commit 67e32ff8 (2015-11-02): FileUtils.copy now filters out source control folders. Also, dir is checked for existence before creating it. Fixes #4034