arduino/Arduino #4036: problem with installing LIFA for Arduino UNO R3

amkamyab, 2015-10-28:
hello
when i try to install  LIFA ( labview interface for Arduino), a message appear and my labview cant connect to Arduino.
1 - i installed LIFA on labview over VIPM
this is message:

Arduino: 1.6.6 Hourly Build 2015/10/27 05:42 (Windows 8.1), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\LIFA_Base\LIFA_Base.ino:26:67: fatal error: OneWire.h: No such file or directory

 #include <OneWire.h>                //For use with OneWire devices

                                                                   ^

compilation terminated.

exit status 1
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

facchinm, 2015-10-28:
Hi @amkamyab ,
you simply need to install OneWire library by Paul Stroffegen; since you are using 1.6.6 you can download it via Library Manager :smile:

amkamyab, 2015-10-28:
thanks for your reply. i will do it and let you warn about it.

facchinm, 2015-11-02:
Hi @amkamyab , any news? Did installing the library solve the issue?

amkamyab, 2015-11-02:
hello
i used arduino version 1.0.5 and it solved problem. it is look like that new version i downloaded didn't match with my arduino.
any way thanks for your help.

closed 2015-11-02 by facchinm