arduino/Arduino #4037: Incorrect documentation for platform.txt pre- and post- build hooks

jamesfowkes, 2015-10-28:
On the wiki: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification

The pre- and post- build hooks are listed as:
```
recipe.hooks.sketch.prebuild.NUMBER.pattern (called before sketch compilation)
recipe.hooks.sketch.postbuild.NUMBER.pattern (called after sketch compilation)
recipe.hooks.libraries.prebuild.NUMBER.pattern (called before libraries compilation)
recipe.hooks.libraries.postbuild.NUMBER.pattern (called after libraries compilation)
recipe.hooks.core.prebuild.NUMBER.pattern (called before core compilation)
recipe.hooks.core.postbuild.NUMBER.pattern (called after core compilation)
recipe.hooks.linking.prebuild.NUMBER.pattern (called before linking)
recipe.hooks.linking.postbuild.NUMBER.pattern (called after linking)
recipe.hooks.objcopy.prebuild.NUMBER.pattern (called before objcopy recipes execution)
recipe.hooks.objcopy.postbuild.NUMBER.pattern (called after objcopy recipes execution)
recipe.hooks.savehex.prebuild.NUMBER.pattern (called before savehex recipe execution)
recipe.hooks.savehex.postbuild.NUMBER.pattern (called after savehex recipe execution)
```
For linking, objcopy and savehex, these are incorrect. The software currently recognises:

```
recipe.hooks.linking.prelinking.NUMBER.pattern (called before linking)
recipe.hooks.linking.postlinking.NUMBER.pattern (called after linking)
recipe.hooks.objcopy.preobjcopy.NUMBER.pattern (called before objcopy recipes execution)
recipe.hooks.objcopy.postobjcopy.NUMBER.pattern (called after objcopy recipes execution)
recipe.hooks.savehex.presavehex.NUMBER.pattern (called before savehex recipe execution)
recipe.hooks.savehex.postsavehex.NUMBER.pattern (called after savehex recipe execution)
```
It looks like I can edit the wiki myself, but I'm not sure if that's the correct thing to do. I'm new to this.

ffissore, 2015-10-28:
Thanks for the heads up: yes you can edit it your self and yes it's the correct thing to do. Keep us posted when you're done :+1:

jamesfowkes, 2015-10-28:
Cool, that's done.

ffissore, 2015-10-28:
Great! Thank you! :)

closed 2015-10-28 by ffissore