arduino/Arduino #4040: Remove RobotIRremote library

PaoloP74, 2015-10-29:
The RobotIRremote library continues to create problems with the installation of the homonymous library irRemote.
Can you delete from installation of IDE 1.6.6 the RobotIRremote and make it optional?
It would be downloaded only by those who own the Arduino robot.

"Translated with Google" :-/

ffissore, 2015-10-29:
This should be fixed in recent IDEs. Can you give some evidence of such problems?

PaoloP74, 2015-10-29:
Recently --> http://forum.arduino.cc/index.php?topic=356077.0
IDE select the wrong library
> Multiple libraries were found for "IRremote.h"
> Used: C:\Program Files (x86)\Arduino\libraries\RobotIRremote
> Not used: C:\Users\Lorenzo\Documents\Arduino\libraries\Arduino-IRremote-master
> Errore durante la compilazione

q2dg, 2015-10-29:
Well, in hourly build from 29-10-2015 2:26  "Esplora", "Robot Control", "Robot IR Remote", "Robot Motor" and "TFT" libraries appear installed by default. I think there libraries should be removed to Library Manager.

PaoloP74, 2016-02-25:
Up!

per1234, 2016-02-25:
If you change the folder name from C:\Users\Lorenzo\Documents\Arduino\libraries\Arduino-IRremote-master to C:\Users\Lorenzo\Documents\Arduino\libraries\IRremote then the IDE will select the correct library because the include filename matches the folder. If you use Library Manager to install IRremote then it will automatically have the correct folder name. However, it seems only logical that if multiple files that match the include are found and none of them are in a folder that matches the filename then the normal include priorities should apply. In this case the file in the sketchbook folder should have higher priority than the one in the IDE install folder but this is not the case in your example. I seem to remember this idea already having been discussed in another issue and rejected. I did notice that if you do:
```
#include <IRremoteInt.h>
```
you get(using hourly build):
```
Multiple libraries were found for "IRremoteInt.h"
 Used: E:\Stuff\misc\electronics\arduino\libraries\IRremote
 Not used: C:\Program Files (x86)\arduino-nightly\libraries\RobotIRremote
```
or with your folder name:
```
Multiple libraries were found for "IRremoteInt.h"
 Used: E:\Stuff\misc\electronics\arduino\libraries\Arduino-IRremote-master
 Not used: C:\Program Files (x86)\arduino-nightly\libraries\RobotIRremote
```
So it's fairly random which library is used when the include filename doesn't match any library folder name.

PaoloP74, 2017-07-18:
Can you delete from installation of next IDE 1.8.4 the Robot IR remote?

cmaglie, 2017-08-18:
I cannot remove it from the bundle (for UX reasons) but I've renamed all the includes from `*IRremote.h` to `Robot*IRRemote.h`, this should fix all the conflicts once for all.

closed 2017-08-18 by cmaglie