arduino/Arduino #4041: User names with accents not correctly supported in Arduino IDE on Windows

noeldev, 2015-10-29:
The "Sketchbook location" and preferences.txt file paths are invalid in the Arduino IDE, File > Preferences properties if the user name contains accents or dieresis; such characters are replaced by other signs for example a Per Mille sign (‰). This also lead to the creation of folders with the wrong names on the disk.

ffissore, 2015-10-29:
Which os are you using? Can you tell the actual location of your sketchbook so we may reproduce your issue?

ffissore, 2015-10-29:
This should be fixed in the current hourly build http://www.arduino.cc/en/Main/Software#hourly. Can you please give it a try?

noeldev, 2015-10-29:
The OS is Windows 10 Pro, x64. The default location of the Sketchbook was "C:\Users\Noël\Documents" which appeared as "C:\Users\No‰l\Documents" in the IDE and a "C:\Users\No‰l" folder was created which includes sub-folders: AppData\Roaming\Arduino15 with 4 files including preferences.txt. I will give the hourly build a try, I will uninstall the current version 1.6.5 and delete the invalid folders first. Thanks.

ffissore, 2015-10-29:
Yeah, it looks a lot like #3360 which was fixed

noeldev, 2015-10-29:
Indeed, this issue is fixed in the latest hourly build: the paths look ok in the IDE and the correct folders are used on the disk, no more No‰l folders. Thanks.

closed 2015-10-29 by ffissore