arduino/Arduino #4043: The Arduino IDE won't open on Windows without administrator rights

noeldev, 2015-10-29:
I have to run the Arduino IDE with administrator rights (right-click on Arduino icon on desktop then Run as administrator in context menu) otherwise the IDE won't open, only the splash screen will show. I could not find this as a requirement so this possibly is an issue. System is Windows 10, x64.

matthijskooijman, 2015-10-29:
I do not believe this is normally a requirement, though perhaps things changed in Windows 10.

One possibility would be that you opened as Administrator some time in the past, and now some files are only accessible by Administrator, so running as a normal user won't work anymore (just guessing here).

Could you try running arduino_debug.exe (possible from a command prompt), to see if that offers some extra debug output or error message?

ffissore, 2015-10-29:
Indeed that's definitely the case. No administrator rights are required

noeldev, 2015-10-29:
I already uninstalled build 1.6.5 and this seems to be fixed with the latest hourly build. I first tried to run arduino_debug.exe from a Command prompt, the IDE did open and I only got these few lines in the Command window:

    Loading configuration...
    Initializing packages...
    Preparing boards...
    Starting...
    Using proxy DIRECT
    Using proxy DIRECT
    Using proxy DIRECT

Then I ran arduino.exe from the Explorer and the IDE did open without the administrator rights. Same with a shortcut to arduino.exe on the desktop. Thanks.

ffissore, 2015-10-29:
Thanks for the feedback!

closed 2015-10-29 by ffissore