arduino/Arduino #4045: Windows - Arduino IDE crashes if the user's home folder contains accented characters

goncalor, 2015-10-29:
When I run `arduino_debug.exe` I get the following.

<pre>
Loading configuration...
Initializing packages...
Preparing boards...
java.io.FileNotFoundException: C:\Users\<b>Gon?alo Ribeiro</b>\AppData\Roaming\Arduino15\preferences.txt (Access is denied)
	at java.io.FileOutputStream.open0(Native Method)
	at java.io.FileOutputStream.open(FileOutputStream.java:270)
	at java.io.FileOutputStream.<init>(FileOutputStream.java:213)
	at java.io.FileOutputStream.<init>(FileOutputStream.java:162)
	at processing.app.legacy.PApplet.createWriter(PApplet.java:574)
	at processing.app.PreferencesData.save(PreferencesData.java:115)
	at processing.app.Base.<init>(Base.java:343)
	at processing.app.Base.guardedMain(Base.java:226)
	at processing.app.Base.main(Base.java:137)
java.lang.RuntimeException: Couldn't create a writer for C:\Users\<b>Gon?alo Ribeiro</b>\AppData\Roaming\Arduino15\preferences.txt
	at processing.app.legacy.PApplet.createWriter(PApplet.java:586)
	at processing.app.PreferencesData.save(PreferencesData.java:115)
	at processing.app.Base.<init>(Base.java:343)
	at processing.app.Base.guardedMain(Base.java:226)
	at processing.app.Base.main(Base.java:137)
</pre>

As you can see the path for preferences has a problem. It should read

<pre>
C:\Users\<b>Gonçalo Ribeiro</b>\AppData\Roaming\Arduino15\preferences.txt
</pre>

The only way I can run Arduino IDE without crashing is by running it as admin. Arduino then creates a folder called

```
C:\Users\Gon‡alo Ribeiro
```

matthijskooijman, 2015-10-29:
Did you add the \<b>\</b> tags in that output?

ffissore, 2015-10-29:
This should be fixed in the current hourly build http://www.arduino.cc/en/Main/Software#hourly. Can you please give it a try?

goncalor, 2015-10-30:
@matthijskooijman, yeah I added the <b> </b> tags.

@ffissore The hourly build opens just fine. I think it is solved.

closed 2015-10-30 by ffissore

goncalor, 2015-11-01:
You're welcome.