arduino/Arduino #4050: Sketch local libraries?!

lmihalkovic, 2015-10-30:
Having run into the issue myself, I noticed that a number of people were asking about libraries inside a sketch project. This is the result of a couple of local changes to the latest IDE.

[image]

Sketches can have a local "libraries" sub-folder with libraries. The outline view displays them in a list. The source files for local libraries can be edited. When selecting a file from a library, all the tabs of the main editor are replaced with equivalent tabs for each of the library's file (doing it this way makes it possible to reuse all the existing Editor logic with very minimal changes).

Would there be any interest for folding this into the release?

VogonJeltz, 2015-11-01:
excellent work, highly appreciated !
+1
(y)

lmihalkovic, 2015-11-04:
For the time being I implemented my own builder as I was wasting too much time on arduino-builder (*go is a no-go* ;-) ). On the bright side, switching between the 3 builders I have is a config option. 
What I really want to look more closely at is the user configurable library-priority scheme I described in #4064 and #4065.

ffissore, 2015-11-16:
Duplicate of #314. See https://github.com/arduino/Arduino/issues/314#issuecomment-143208797

closed 2015-11-16 by ffissore

lmihalkovic, 2015-11-17:
> What change would like to see?
>
> The ability to include libraries on a per-sketch basis. This should probably be in the code/ folder to mirror Processing, or a libraries/ folder (to mirror other library locations). In the long run, this should support pre-compiled libraries too (issue #199), but source libraries would be a good start.

> Why?
> 
> So people can include all the dependencies for a sketch in one place. This would also be a nice place to put assembly files for a sketch (since you can't edit them in the IDE) - issue #274.

After 3 years I have to assume that you actually see how much good it does to the codebase as well as the number of other doors it would open for users, and accept that ArduinoIDE is in sunset mode, destined to receive only the minimum amount of maintenance until something else replaces it. no pb.

VogonJeltz, 2015-11-17:
just curious about how the Arduino Studio actually handles the libraries issue...
http://labs.arduino.org/Arduino%20Studio

lmihalkovic, 2015-11-17:
interesting question... I sure like what they call "inline Quick Guide".. I imagine that Arduino Create will come up with something equally compelling. 

Nonetheless in the meantime I would love to plug the gap for IDE... at the moment all I can do is open the complete Reference page for a library in a internal simple web browser I wrote (see #4069) from the Outline view. But yesterday I understood the PdeKeywd parseHTMLRef.. so I will add linking to more contents from the quick access results and from a keyword in the editor. Still, full inline help at the caret position wins the day.

btw if someone at Arduino.cc looks at the log files, "ArduinoIDE-1.7.0" is my browser. Would be nice if there was a simple way to access the help contents without the full web site header, but if that does not become available, I'll have to add a JS filter to the browser so strip it.